emoncms/emoncms #1214: Issues with UTF-8 Encoded parameters

borpin, 2019-02-24:
Ref recent issues with EmonESP which appear to be sending UTF-8 encoded parameters in the URL. e.g. https://community.openenergymonitor.org/t/emonesp-does-not-update-emoncms-debian-and-ubuntu-apache-error-400/10218

I wonder if there should be a URL decode on this line - belt and braces?

https://github.com/emoncms/emoncms/blob/fefeef34278834f33359213aed3b9bbbd4d1ab90/param.php#L80

TrystanLea, 2019-07-11:
Thanks @borpin, noted.

TrystanLea, 2020-03-19:
I think this is not something we can sort on the emoncms side, it requires that the device posting data encodes the data string in the URL correctly. Please re-open if you think this is still an issue.

closed 2020-03-19 by TrystanLea