emoncms/emoncms #1216: Server info conflicting info for git version

pb66, 2019-02-26:
Seems the git sourced version (9.9.4) info is lagging behind the emoncms inbuilt version (9.9.7-beta)

[image]

Is this dependent on someone always remembering to do something in addition to updating the version.txt and creating the release tag?

pb66, 2019-02-26:
Answering my own question, yes it does. I see @TrystanLea  has just posted about adding the tag annotation on the forum https://community.openenergymonitor.org/t/announcement-of-9-9-6/10067/14?u=pb66

[image]

pb66, 2019-02-26:
Is there supposed to be a space after the "9.9.7-1g" ? The "ae268e04" part matches the commit number and refers to a point that can be found, or am I reading it wrong?

borpin, 2019-02-26:
I think that, because you are on master and the tag should point to a commit on stable, I'm not sure it will ever look right on master unless a 'beta' tag is created for the commit that changes the version file to 'beta'.

The hyphen tells you the relative position of your HEAD to the last annotated tag.  The number is the number of commits, but I am not sure what the g means - possibly because it is on a different branch?

This is all. I think, because master is not used for releases.

closed 2019-04-30 by glynhudson