emoncms/emoncms #1226: can not source feed a real sensor. Cannot use virtual feed as source

danpadure, 2019-03-26:
When trying to create new virtual feed from a real one using source feed, emoncms says:
ERROR: Could not save processlist. Cannot use virtual feed as source

I tried multiple times.
There is also a tread on the forum
https://community.openenergymonitor.org/t/virtual-feed-add-values-from-input/1835

latest version, 9.9.8

danpadure, 2019-03-26:
[image]

alexandrecuer, 2019-03-26:
true it is not working whith emoncms last version, but you can use the postprocess module...

I think postreating feeds can be a very nice alternative....if you install the last module of postprocess you will find a simple formula algorithm and many others tools
https://github.com/emoncms/postprocess
try it with a git fetch/git pull

regards

Alex

danpadure, 2019-03-26:
yes, sounds like a great module, but...

it doesn't work either
I did find in apache error.log this:
PHP Notice:  Trying to get property of non-object in /home/pi/postprocess/postprocess-module/postprocess_model.php on line 47, referer: http://192.168.12.42/emoncms/postprocess

PHP Fatal error:  Uncaught Error: Call to a member function bind_param() on boolean in /home/pi/postprocess/postprocess-module/postprocess_model.php:33\nStack trace:\n#0 /home/pi/postprocess/postprocess-module/postprocess_controller.php(134): PostProcess->set(1, '[]')\n#1 /var/www/emoncms/core.php(64): postprocess_controller()\n#2 /var/www/emoncms/index.php(210): controller('postprocess')\n#3 {main}\n  thrown in /home/pi/postprocess/postprocess-module/postprocess_model.php on line 33, referer: http://192.168.12.42/emoncms/postprocess

should I open an issue in postprocess github or is related to the main emoncms problem?

danpadure, 2019-03-26:
I forgot to mention: I see the new feed in feeds menu. It's NULL

And in postprocessor I do not see it in my processes

alexandrecuer, 2019-03-26:
Do you run the postprocess version with the log view , as on the image below ?

[image]

I am running emoncms 9.9.6 (9.9.7beta) on most of my PIs, and postprocess is running very fine

I will try to catch up master and to reproduce your bug.....

danpadure, 2019-03-26:
my module looks a lot more empty  :P

[image]

alexandrecuer, 2019-03-26:
OK you've got the previous version of postprocess

I've joined the master version and postprocess is working fine for me : I cannot reproduce your issue....I had successfully created a new process :

[image]

I can see your process list is empty....is it impossible for yu to create a new process ?
take a simple watt feed coming from an emonTX and try to rebuild the accumulated feed with the powertokwh algorithm.....

danpadure, 2019-03-27:
I can create a new process. but it does not work and does not show up in the process app
only in fees I can see it but it's NULL

[image]

Maybe it's something wrong with my opencms.  
Also backup app looks weird and I can't see nothing on the black screen.  Backup does not run when I press the button.
But I can run the backup form the command line emoncms-export.sh  and then I can see the archive it in the backup app

[image]

danpadure, 2019-03-27:
what do you mean I have the "old" version of postprocess?  How should I get the new one?.

danpadure, 2019-03-27:
ok I downloaded the latest version

But still I see nothing in "my processes"

How can I enable logging?

danpadure, 2019-03-27:
ok, I did an update to emonpi, emonhub, and all update scritps I could find
and now I have some "progress"  meaning I can see the processes I create.   I see them but still they won't run

So basically neither the run process nor the backup buttons do anything
somehow I feel the 2 issues are related.

But I can't see anything in any logs, it seems the updates did fix whatever issue was

[image]

danpadure, 2019-03-27:
I manualy ran sudo php /home/pi/postprocess/postprocess_run.php
and now all processes got updated

I created a new process, clicked "run process" and now I get a NEW error:

[image]

Man, I can't get a brake, can I?  :)

alexandrecuer, 2019-03-27:
You seem to be on the right way
was the log file updated after you launched the php script ? do yo have points in your feed 20 ?

Anyway,, you run a PI zero and most of us are running PI3....I have no experience on PI zero
I think you should open a thread on the open energy monitor forums : some forum users are using PI zero

Alex

danpadure, 2019-03-27:
Yes, I have points in all feeds (I added some more)

PI zero should be similar to pi3
I just had to build my own card, because the oct 2018 one does not see the wifi

Strange thing: I can't replicate the error. I restarted apache2, added a new process, ran again the script.  So the only time I got a feedback from the buttons is when I got the error. Now again nothing works

[image]

alexandrecuer, 2019-03-27:
who is the owner of /home/pi/data/postprocess.log
user pi and group pi ?
it must be a rw file...

it seems you need to go deeper in emoncms settings files

danpadure, 2019-03-28:
file was non existant.  I created it now

-rw-r--r-- 1 pi       pi       0 Mar 28 07:38 postprocess.log

alexandrecuer, 2019-03-28:
OK fine 
is it working better ?
you should do the same with the backup log file

danpadure, 2019-03-28:
nope, not even if I manually run
sudo php /home/pi/postprocess/postprocess_run.php

still:
-rw-r--r-- 1 pi       pi       0 Mar 28 07:38 postprocess.log

But I managed to reproduce the error when clicking "run process" button:
I have to keep pressing the buttons (does not matter which one), and after if I press them a lot of times, after ~5-8 tries I get the error.
But still no process runs

This should be pretty simple: press a button to run something

danpadure, 2019-03-28:
I could make cron run /home/pi/postprocess/postprocess_run.php every minute
but that's just a workaround, not a fix

alexandrecuer, 2019-03-28:
emoncms is using a python service-runner, which is launching the feedwriter and also arbitrary commands from modules like postprocess
so when you create a process or when you click run process, datas are dropped into redis and the service-runner read them and lauch the php script you are manually running....
is the service-runner correctly running on your zero ?

danpadure, 2019-03-28:
pi@emonpi:/var/log $ ps -A | grep service
  360 ?        00:00:16 service-runner

how would I check if it's running fine?

greeebs, 2019-03-29:
If it is the latest service-runner (using systemd and python) as opposed to the one triggered by a crontab entry, you can use:
```
systemctl status service-runner
```
to view the status of the actual service, and
```
journalctl --unit=service-runner
```
to see the last entries from the log. `service-runner` should log to the journal every time it gets a new command to run.

If the `systemctl status` command returns an error such as "not-found", then it is likely you have a crontab initiated service-runner, which may or may not be reading its "flags" from redis.

danpadure, 2019-03-29:
it's not, it's triggered by crontab

danpadure, 2019-03-29:
ok so close to fix:
I installed redis (might not have been installed at all)
I installed the new services: emoncms_mqtt, Feedwriter, Service Runner. Now backup works, buttons from post processor work

however post processor does not seem to run automatically.  I have to manually press the buttons.

I guess it should run automatically?

alexandrecuer, 2019-03-29:
Nice to hear :-)
it is not automated... postprocess_run.php is run when created
when your feeds got updated, you have to update manually....
automation has to be developped..there is a specific issue on that...

alexandrecuer, 2019-03-29:
What is strange is that in your admin view (first screenshot you attached to this thread) redis-server appeared as running.....should not as you installed it further from what I understand....
@emrysr : is it possible that admin module returns redis-server to be running if not installed ?

danpadure, 2019-03-30:
when I thought it was over...
I deleted older test processes
I added new ones.  But I have a problem:  when I create addfeeds, the process appears for 1-2 seconds in the list, then dissapears.  I can see i in feeds, it ran ok, but I can no longer see it in my processes, so I can't update it

If I create other processes, like powertokwh or scalefeed, they work ok and I can see them
I tried basicformula but I don't know the syntax how to creat them

also the  postprocess.log file seems to overwrite itself, I always get only the last output, somewhere there should be a >> instead of >

[image]

[image]

alexandrecuer, 2019-03-30:
To use basic formula, you just have to know the id numbers of the feeds you want to involve in the formula...for exemple if you want to multiply feed 1 by feed 2 and to add  feed3 multiplied by 2, the formula to enter is : 
f1*f2+2*f3
One limitation is that yu cannot use brackets ()

danpadure, 2019-03-30:
I have only 1 field, where do I enter the resulting feed name?

[image]

When I click create, nothing happens, except the create button dissapears

[image]

alexandrecuer, 2019-03-30:
This is a bug : I had this bug once and could not reproduce it

normally you should have a text input for your formula and a text input for the feed name
when both are correctly filled, the submit button appears

in your screenshot, you only have the text field for the output feed name

If I remember well, it disappeared when I emptied the browser cache

danpadure, 2019-03-30:
Indeed, I used another browser and basicformula works fine now
I think I can use this instead of addfeeds, right?  

In the new browser I tested again addfeeds and the process remains in the list until the calculations are done, then 

But if I can use basicformula, it should be fine enough

alexandrecuer, 2019-03-30:
you can use basicformula instead of addfeeds....
It will do the job, even though it is not yet very user friendly as far as you have to know the feeds numbers and cannot yet pick them in a list....

Anyway, what was the browser with which the bug occurred ?

danpadure, 2019-03-30:
it was chrome
that's a strange bug

danpadure, 2019-03-30:
thanks for the help
I'll close the thread for now.
But I'm opening another one  :P

closed 2019-03-30 by danpadure