emoncms/emoncms #1228: error chaging the graph color

danpadure, 2019-03-30:
I'm getting an error trying to change colors:
emoncms error:
message: uncaught typeerror: cannot read property 'lines' of undefined
url:ip/emoncms/Modules/graph/graph.js?v=1
line: 756
column: 25

9.9.8-4-gd0db7a57

[image]

danpadure, 2019-03-30:
oh, this happens when I click on the color in the top left part of the graph

njfaria, 2019-04-07:
with some debug info
[image]

glynhudson, 2019-05-09:
Thanks, we're able to replicate. Will get this fixed very soon

emrysr, 2019-05-09:
You should change the feed's colour from the list below the chart no in the chart legend. I will fix the js error when the colour is clicked.

This could be added as a feature. Please open an issue in the graph module repo to discuss this.
https://github.com/emoncms/graph

closed 2019-05-09 by glynhudson