emoncms/emoncms #1231: Sidebar branch: menu organising

glynhudson, 2019-04-06:
"Setup" is no longer a particularly applicable name to the sidebar. I think we should chance this to "Menu" or "Emoncms"? To be discussed....

Also as @borpin suggested some of the sidebar menu items such as "backup", "WiFi" "admin" could be hidden in a "System" menu. We could use the pop-out tray at the bottom of the sidebar for this system menu since "account" can be accessed via the user profile (the best place IMO). 

See forum discussion: https://community.openenergymonitor.org/t/beta-new-emoncms-sidebar-navigation-layout-feedback-request/10435/33?u=glyn.hudson

emrysr, 2019-04-08:
I like all the above. good thinking. i'll work on them tomorrow

borpin, 2019-04-08:
To `system` I'd add
* emonhub
* Device setup

emrysr, 2019-04-30:
[image]

This is the current menu layout I've set to match the above. Each menu item's position can be controlled by the module's `*_menu.php` file

emrysr, 2019-04-30:
these are the items in the user dropdown:
[image]

borpin, 2019-04-30:
What is groups on the Sidebar?  Is this user management?

If I click on 'Administration', what happens to the sidebar?

borpin, 2019-04-30:
Personally, I think the settings should be part of the sidebar so that all those items listed, simply appear in the sidebar when the cog icon is clicked.

You could then have a sub menu under admin for 'Users', 'Update', 'Logger', 'Server Info' & 'Client Infor` which just move you to that part of the page.

I am presuming clicking on Backup takes me into a backup sidebar still? A very confused UI if it does.

glynhudson, 2019-04-30:
>What is groups on the Sidebar? Is this user management?

It's the groups module https://github.com/emoncms/group

>If I click on 'Administration', what happens to the sidebar?
 
It stays there

[image]

>I am presuming clicking on Backup takes me into a backup sidebar still? A very confused UI if it does.

No, backup has tabs to switch between import/export. Notice the 'cog' is highlighted to indicate that backup is a system/admin module 

[image]

There are lots of ways to structure the menu. Having been testing and using the new sidebar extensively for the past few months I think it's a big improvement. It can always be tweaked over time. Personally I like the segregation of having the system menu on the cog and main menu on the left hand side. It's very similar to the Discourse forum layout

[image]
[image]

closed 2019-04-30 by glynhudson