emoncms/emoncms #1235: Travis Failing & running tests locally

glynhudson, 2019-04-08:
Currently travis builds are failing for Emoncms, investigating this issue it seems that Traivs is failing on a user logout test: https://travis-ci.org/emoncms/emoncms/jobs/516265670#L1164

It's very impressive the test setup @cagabi has built for Emoncms. 

To try and debug this error I've been trying to setup to run the test locally following the instructions in `tests/`. 

I get the following error when running the test, it sounds like ChromeDriver is not communicating with Chrome. 

```
:~/Documents/emoncms/tests$ node_modules/.bin/wdio wdio.conf.jsERROR: unknown error: Chrome failed to start: exited normally
  (unknown error: DevToolsActivePort file doesn't exist)
  (The process started from chrome location /usr/bin/google-chrome is no longer running, so ChromeDriver is assuming that Chrome has crashed.)
  (Driver info: chromedriver=2.43.600233 (523efee95e3d68b8719b3a1c83051aa63aa6b10d),platform=Linux 4.15.0-47-generic x86_64) (WARNING: The server did not provide any stacktrace information)
Command duration or timeout: 3.98 seconds
Build info: version: '3.141.5', revision: 'd54ebd709a', time: '2018-11-06T11:58:47'
System info: host: 'glyn-desktop', ip: '127.0.1.1', os.name: 'Linux', os.arch: 'amd64', os.version: '4.15.0-47-generic', java.version: '1.8.0_101'
Driver info: driver.version: unknown
```

When this error appears I also get lots of tabs opend with paramaters like `http://--disable-background-networking/` and `http://--disable-client-side-phishing-detection/`

cagabi, 2019-04-15:
Hi, sorry just saw this issue.

About running the tests locally, I have just run the tests in my laptop and they passed.

For the time being until we find what the problems are, have you tried to run the tests with Firefox?
`BROWSER=firefox ./node_modules/.bin/wdio wdio.conf.js`

I recently had some problems running the tests for MHEP locally. After a lot of looking around, the problem was the version of node that I had recently upgraded from 8.15 to 11.14. I added some text to MHEP [readme.md](https://github.com/emoncms/MyHomeEnergyPlanner/blob/development/tests/readme.md#set-up) about how to manage with nvm which version of node to use. But my problems were related to node modules so it was more obvious. I don't think this will be the case for you.

After doing some google search, I have seen in several places that they say to add `'--no-sandbox','--disable-dev-shm-usage'` to [wdio.conf.js#L18](https://github.com/emoncms/emoncms/blob/master/tests/wdio.conf.js#L18). Could you try?

Also, have you checked your chrome is installed in /usr/bin/google-chrome?

The [here](http://chromedriver.chromium.org/downloads/version-selection) they say that the version of Chrome and Chromedriver have to match. I didn't know that

cagabi, 2019-04-15:
With regards to the test itself. 
The build you are linking is failing because the sidebar branch has changed the theme and now to logout you first click in the user name and then the logout link. #
The test is trying to click straight away in the logout link (and as the error message says it is not possible because it is not visible). So in that branch they need to change [the code](https://github.com/emoncms/emoncms/blob/sidebar/tests/Lib/emoncms_tests_helper.js#L34) of the test.

For the master branch, it's strange, the test were failling until 6 days ago when they started working again

closed 2019-07-11 by TrystanLea