emoncms/emoncms #1236: empty virtual feed causes error in graph

djh42, 2019-04-09:
If there is a virtual feed with no data (because of the bug with source feeds, it's not possible to set up a processlist), then graph displays:

> Request error
> Warning: end() expects parameter 1 to be array, boolean given in /var/www/emoncms/Modules/feed/feed_model.php on line 588
> false

This appears to be because Modules/feed/engine/VirtualFeed.php::get_data returns false if there is no data, whilst Modules/feed/feed_model.php does not check for that result.

TrystanLea, 2019-07-11:
@chaveiro would you be able to help with a solution to this?

TrystanLea, 2020-01-20:
Could you provide more details on how to replicate this issue, testing here and an empty source feed seems to result in a zero value virtual feed but no error.

djh42, 2020-01-20:
1. Go to the feed list
2. Click on New virtual feed
3. Click on Save (virtual feed is created)
4. Click on Setup menu then Graphs (empty graph is displayed)
5. Click on feed list
6. Select Virtual then click in left box of New Virtual Feed
(Error message appears)

It may be worth noting that I'm running low-write 9.9.8

TrystanLea, 2020-03-27:
Thanks I was able to replicate and have now merged a fix into the master branch:
https://github.com/emoncms/emoncms/commit/d5fa8eb7e1ba1f0bb2acad17f32f59a0d95823ae

should be in stable soon

closed 2020-03-27 by TrystanLea