emoncms/emoncms #1239: Update travis tests on sidebar branch to correctly logout

glynhudson, 2019-04-16:
The Travis test builds are failing because the sidebar branch has changed the theme and now to logout you first click in the user name and then the logout link. #
The test is trying to click straight away in the logout link (and as the error message says it is not possible because it is not visible). So in that branch they need to change [the code](https://github.com/emoncms/emoncms/blob/sidebar/tests/Lib/emoncms_tests_helper.js#L34) of the test.

See :https://github.com/emoncms/emoncms/issues/1235#issuecomment-483237407

emrysr, 2019-04-30:
I've direclty edited the branch within github.
```
browser.click('#user-dropdown')
browser.click('#logout-link');
```

I've also updated my `feature/issue-1194-new-sidebar` branch to include this change.

glynhudson, 2019-04-30:
Nice, the logout seems to be working now. But it still seems to be failing on another element?

emrysr, 2019-05-09:
@glynhudson Is this still failing? can this be closed?

glynhudson, 2019-05-09:
It's still failing on almost every build. It's hard to figure out what's now causing the failur, see https://travis-ci.org/emoncms/emoncms/builds

[image]

TrystanLea, 2019-09-03:
Is this still an issue?

alexandrecuer, 2019-09-06:
I dont think it is really an issue as the EmonScripts are available....you should remove the Travis icon, it should be cleaner....
What was initially the need for such a complex travis.yml file  ?

glynhudson, 2019-09-06:
There is no need for such a complex travis file, it's obviously causing more issues than its solving and the failed build badge on the repo front page looks bad. I vote for reverting to a simple travis file to just check the PHP syntax like we used to have or just removing it totally if it can't be made to work. Do you agree @emrysr?

alexandrecuer, 2019-09-07:
So I propose a very simple .travis.yml file :
```
language: php
php:
    - '5.6'
    - '7.2'
    - '7.3'
install:
   - composer install
script: 
  - composer test
```
It will check the PHP syntax only and the build will come green again
Do you want a PR in that sense ?

glynhudson, 2019-09-07:
Yes, that would be great. Thanks

closed 2019-09-07 by glynhudson

emrysr, 2019-09-09:
@glynhudson - I agree this simpler test now passes - we weren't making use of the other tests. Thanks @alexandrecuer looks much better now
[image]