emoncms/emoncms #1245: Sidebar tab navigation shotcuts

glynhudson, 2019-04-30:
It would be useful to be able to navigate the new sidebar using tab navigation. 

In my option tabs should first traverse the left side bar, bookmarks, bookmark star, then the system cog menu 

In Emoncms V9 it was possible to tab though the top bar then the side bar. I found this quite useful

glynhudson, 2019-05-15:
That's great, the tabbed nav works exactly works perfectly. This is exactly what I had in mind. 

However, after checking out this version the System cog menu now overflows the side of the screen, see:

[image]
[image]

glynhudson, 2019-05-15:
Same issue with the Profile menu

emrysr, 2019-05-15:
ah! well spotted.

emrysr, 2019-05-15:
fixed in the pull request - https://github.com/emoncms/emoncms/pull/1266

[image]

PR #1266 (2019-05-15): Feature/issue 1245 tab nav
fix #1245 

the hidden sidebar elements are now not accessible via the keyboard, therefore the user can "get to" all the relevant menu items via the keyboard.

the "top navbar menus" are the first in the "tab order" - so it takes 6 presses of the `tab` key before the sidebar menu items are accessed.

[screenshot with item tabindex overlayed]
[image]
closed 2019-05-15 by glynhudson