emoncms/emoncms #1256: Readme Module installation- should use stable branch

whatsupskip, 2019-05-03:
Currently the documentation says

> To install Apps, cd to the Modules folder, and git clone 'apps':
> 
> cd /var/www/emoncms/Modules && git clone https://github.com/emoncms/app.git

However this has caused problems with the version 9.9.9 and for [others previously](https://community.openenergymonitor.org/t/problem-with-the-basic-dashboard-data/8503/5).

The documentation should be changed to 
> To install Apps, cd to the Modules folder, and git clone 'apps':
> 
> cd /var/www/emoncms/Modules && git clone -b stable --single-branch https://github.com/emoncms/dashboard.git

glynhudson, 2019-05-03:
Thanks, good idea, docs have been amended.

closed 2019-05-03 by glynhudson

whatsupskip, 2019-05-03:
Great.

commit 50fd8028 (2019-05-03): clone stable module fix #1256 & remove PHP5