emoncms/emoncms #1258: Error with load_language_files with update 9.9.9

BB-Rabbit, 2019-05-05:
Hi all,

Since I have upgraded my version of Emoncms from 9.9.8 to 9.9.9 I am unable to load my Emoncms web GUI. I get the following error:

Fatal error: Call to undefined function load_language_files() in /var/www/emoncms/Modules/dashboard/Views/dashboard_view.php on line 15

If I comment out this line in the dashboard_view.php file, then I can view my database, but I get other errors, which I can click past to see my data.

I hope I have not just done something stupid here! Thanks in advance.

glynhudson, 2019-05-08:
Hi since, you're running Emoncms 9.9.9 you must be running Emoncms 'Stable' branch. In this case please also run the stable branch of the modules e.g

```
cd /var/www/emoncms/dashboard
git checkout stable
git pull origin stable
```

Or alternately you could move to running master branch of Emoncms core, you should then ensure all other modules are also running master.

BB-Rabbit, 2019-05-08:
Thanks Glyn, That fixed my problem thanks very much. Is there a quick way to check to see if my modules/core are all linked to stable?

I have never come across this issue before, as I usually just run the update.sh script and it updates everything for me.

glynhudson, 2019-05-09:
Great. 

If you run an emonPi update the current branch for each module will be displayed in the update log. The default for emonSD is stable for all modules that have a stable branch. 

This is just a temporary issue, while Emoncms V10 has been just released to master and not yet stable branch. Hopefully it will be released to stable next week and this issue should not be a poblem. However I would always recommend running the same branch to avoid future issues.

closed 2019-05-09 by glynhudson