emoncms/emoncms #1268: password w/ special characters problem since commit 96b5ae0

Chrischi-, 2019-05-14:
It looks like passwords with special characters at the end (e.g.: #) will not work anymore since commit 96b5ae0.

I had to use `resetpassword.php` to set a new one without special characters.

glynhudson, 2019-05-15:
Oh dear, very sorry about this. Thanks a lot for reporting. We will investigate immediately

emrysr, 2019-05-15:
this was not changed at the same time as the above commit. the changes shown there are javascript changes and not related to sanitizing user inputs...

I am unsure without further investigation to identify when this error crept in...
https://github.com/emoncms/emoncms/blame/b00eb94ed9aa3a4ce3b96427d68a6d3bbf43e518/core.php#L88

emrysr, 2019-05-15:
https://github.com/emoncms/emoncms/pull/1269

Chrischi-, 2019-05-15:
I have been using a password with special characters for a long time. If I go back to the commit, the login works again. From this commit on there are the mentioned problems. However, I have not tested any intermediate commits.

glynhudson, 2019-05-15:
@Chrischi- this issue should be fixed now https://github.com/emoncms/emoncms/pull/1269 has been merged. Please could you confirm this fixes the issue for you?

Chrischi-, 2019-05-15:
@glynhudson tested and works!

@emrysr thanks for the fast "fix"

glynhudson, 2019-05-15:
Great, thanks for testing 👍 

Nice work @emrysr

closed 2019-05-15 by glynhudson