emoncms/emoncms #1274: If no Feeds are setup Input page should be displayed by default

glynhudson, 2019-05-16:
Currently, the Feed page is displayed by default after login. When setting up a new system the Feed page will be empty therefore it would be more user friendly to display the Input page

emrysr, 2019-05-20:
the default view is in the config file:
https://github.com/emoncms/emoncms/blob/4b7ec56f51f3d0e51ae18b2f0c1c893b939d3d08/default.settings.php#L129-L130

@glynhudson we could change the default setting (above) ? or we could change `feeds` controller to check for **"no feeds"** before re-directing to `inputs` if empty

emrysr, 2019-05-20:
[image]
@glynhudson - there is already a js check for no feeds... should I change this to redirect the user to the input list  ?`input/view`

glynhudson, 2019-05-22:
> there is already a js check for no feeds... should I change this to redirect the user to the input list 
 input/view

Mmm actually maybe it's a good idea for users to see this message to understand what Feeds are before going to the Inputs. I guess if there is a danger if a new user goes straight to Inputs they may think their data is already being logged if visible in Inputs. Best leave it how it is for now. 

This issue when I tested was this helper message was not displayed, I will do some more testing an open new issue if required.

closed 2019-05-22 by glynhudson