emoncms/emoncms #1277: Bookmarking Feed page clears all bookmarks

glynhudson, 2019-05-18:
Bookmarking Feed page seems to turn off Device Module beta setting and clears all other bookmarks. See this screen cast video when I managed to re-create the issue twice in a row: https://drive.google.com/file/d/1I-cAJL4v6Di6PHjda9FHwUZrp_pr43Ql/view

This is a minor issue since I can't think of any sane reason why anyone would want to bookmark the Feed page since it's easily accessible via the same side menu. Maybe it should not be allowed to bookmark the Feed and Input page if this makes it easier to fix.

Device module beta should be turned on by default and maybe the beta toggle taken away. What's the best way to enable Device Module beta by default Will discuss this with @TrystanLea

emrysr, 2019-05-20:
fixed now https://github.com/emoncms/emoncms/pull/1278

emrysr, 2019-05-20:
https://github.com/emoncms/emoncms/pull/1279

closed 2019-05-21 by glynhudson