emoncms/emoncms #1291: New device added by Open-Red but stays inactive

MrHappy, 2019-05-24:
Have the latest stable-branch installations, most recent node-red en a flow for openweathermap.
I also have a flow for nest and that works fine (created in emoncms v9).
The openweathermap-flow sends data to emoncms, but the device stays inactive.
Although I can create a feed from the data also the feeds stay inactive and no data get recorded (nor shown in the graphs).
Am I missing something or is there a small flaw in the new version?

TrystanLea, 2019-07-11:
Hello @MrHappy sorry to miss your post, did you resolve this issue? It sounds like a MQTT related issue?

MrHappy, 2019-07-11:
I did resolve it. Unfortunately I don't exactly recall how however... I believe it was something in Open-Red, I recreated the flow in other way and then it started working...

closed 2019-07-11 by MrHappy