emoncms/emoncms #1292: Powered by bar scrolls with page

MrHappy, 2019-05-25:
The 'Powered by bar' scrolls with the page, thus hovering over my graphs instead of staying on the bottom of the screen.
This happens as well in Firefox as in Edge.
I'm using EmonCMS 10.0.2.

glynhudson, 2019-05-25:
Thanks for reporting. Does this happen after clearing browser cache or using private browsing window? Do you see the same effect on Chrome. Please post a screenshot if possible. I cannot yet replicate the issue. We will try and investigate

MrHappy, 2019-05-26:
[image]
A screemshot from the page on my phone (Chrome), also tried it in private-tab with same result...

closed 2019-05-26 by glynhudson

glynhudson, 2019-05-26:
@MrHappy fix has been merged into Master branch. Please could you test and confirm the issue has been fixed for you by checking out the Master branch and pulling in the latest changes

MrHappy, 2019-05-26:
Added the master-branch to a subdirectory under my stable-installation and it works fine.

PR #1293 (2019-05-26): dashboard height fix
fix #1292 

footer now fixed at bottom of the content..

[image]