emoncms/emoncms #1294: Flexible log_location for emoncms

TrystanLea, 2019-05-27:
At present the emoncms.log is located in /var/log/emoncms.log and the emonpiupdate.log, emoncms-sync.log, backup.log, postprocess.log are all located in /home/pi/data/ e.g /home/pi/data/emonpiupdate.log.

A more consistent approach without creating logfile location settings for every since logfile would be to specify an emoncms log folder, e.g: /var/log/emon

I have implemented this approach in a development branch here https://github.com/emoncms/emoncms/tree/log_location.

The existing emonSD images would need an rc.local entry to create the /var/log/emon folder and set the ownership to pi, as well as the creation of the emoncms.log in /var/log/emon with 666 permissions. 

The new install script can create the /var/log/emon folder and emoncms.log with 666 permissions as part of the install procedure.

TrystanLea, 2019-05-27:
Here's a direct comparison showing the proposed changes:
https://github.com/emoncms/emoncms/compare/log_location

TrystanLea, 2019-05-27:
There would of course be corresponding changes to the emoncms sync, postprocess & backup modules.

TrystanLea, 2019-05-27:
Changing the log location to /var/log/emon on the current emonpi image would mean the emonpiupdate.log would not persist as it would be in ram. If we feel persistence is needed we could symlink this log from /home/pi/data/emonpiupdate.log

pb66, 2019-05-27:
I have commented in the forum threads about possibly changing the service-runner (also mqtt_emoncms and feedwriter) service(s) to create their own log folder as suggested for emonhub, this would negate any need for an entry in rc.local, be more robust and simplify the updater/install scripts to.

https://community.openenergymonitor.org/t/emonsd-issues-created-using-new-scripts/10918/60?u=pb66
https://community.openenergymonitor.org/t/var-log-still-filling-up/10921/27?u=pb66

closed 2019-06-22 by TrystanLea