emoncms/emoncms #1309: logrotate install - No check made on previous install

borpin, 2019-06-12:
https://github.com/emoncms/emoncms/blob/9ecf60c66bf3add6771de37d45b4568a4e70b908/scripts/logger/install.sh#L11

The script assumes it is being run on an unmodifed system.  A check needs to be made to see if the logrotate system has actually been modified so there is no error.

Discussion https://community.openenergymonitor.org/t/error-in-log-on-update-to-10-1-0/11165

pb66, 2019-06-12:
see https://github.com/emoncms/emoncms/commit/24dbaf44b16c2b28a0a92ab105e977786bac8f69 I've already brought this up as the current setup will simply overwrite any original backed up files on each run of the updater. Either we don't ever back them up or we back them up just once, we can't back them up at every update as it is pointless if the "backups" are just copies of the current repo file. My solution will only backup real files, it will ignore the symlinked replacement files.

borpin, 2019-06-13:
Certainly connected issues.

Why is the logrotate install script in the main repo anyway?

TrystanLea, 2019-11-25:
Closing as this has now changed with the new build script.

closed 2019-11-25 by TrystanLea