emoncms/emoncms #1325: Faster column size allocation

TrystanLea, 2019-06-22:
The dom based column size allocation method is becoming slow on input pages with large amounts of inputs. Searching through character length seems to work well, what do you think @emrysr https://github.com/emoncms/emoncms/commit/409c2ba3d5a3cea8099e0e4160ac4c06f094c553

emrysr, 2019-06-22:
Character length can be tricky on a variable width fonts (proportional vs monospaced) as not all the characters will be the same width.

Should be ok if we allow for an offset or padding to be added to the charter length

TrystanLea, 2019-06-22:
great, can we just account for the character length based on a maximum character width? I've also added 20px of padding.

I've been testing the interface with the firefox JS flame chart and it highlights that a lot of time is also taken by the moment calls for the input tooltips, onresize and a few other bits. I think we could get it running much faster. Perhaps moving the width calculation pre draw as well so that the width changes are not applied after drawing the table the dom..

emrysr, 2019-06-22:
The column width calculations are done once the new data is loaded or when the browser window resizes. I'll look at this next and post a pull request for you to review.

TrystanLea, 2019-06-22:
Great, have a look at my last 3 commits, I've got the column width calculation working, but it would be great to see what else we can do to improve it further.

In particular commit: https://github.com/emoncms/emoncms/commit/409c2ba3d5a3cea8099e0e4160ac4c06f094c553

closed 2019-07-03 by TrystanLea

xekon, 2020-01-02:
I tried this out hoping it would solve the values column from being chopped down, inspecting the element shows:

```
<span data-col="D" class="value text-center" style="width: 33px;">
                        118.2
                    </span>
```

[image]