emoncms/emoncms #1336: Ensuring consistent code style

reedy, 2019-06-28:
I don't mind how code is styled (things like tabs vs spaces - #131), but it's nice if it's standardised, and also enforced by CI to prevent regressions and keep things from regressing

Is anyone against using PHPCS on this repo, and gradually enabling rules to define a coding standard, to make things consistent?

Something that does slightly concern me is the lack of braces being used on ifs... And hopefully we're all aware of the [problems](https://embeddedgurus.com/barr-code/2014/03/apples-gotofail-ssl-security-bug-was-easily-preventable/) this can cause elsewhere...

It'll be a bit disruptive to update the various files (that aren't 3rd party) to consistently use braces, but I think the benefit in the long run is worth it. Most rebasing should be relatively trivial

TrystanLea, 2019-06-29:
Thanks @reedy Im in the process of working through bringing two forks of emoncms together, the code base for emoncms.org and the main repo here, Id like to avoid large styling changes until that is done. 

Im happy to standardise on including braces, I've generally adhered to this but I know others favour missing them out where not strictly needed. Happy to receive pull requests with this change specifically.

reedy, 2019-06-29:
That's fair enough, I'm in no rush to do so, but I know from other experience it'll be worth it in the long run for readability and alike. Certainly other stuff that can be done, such as working on the subproject repos ala https://github.com/emoncms/wifi/pull/24 as well as cleanup here in the form of #1333, #1334 and #1335 to fix up various other issues... Stuff like this will probably keep me busy for a while, getting used to the codebases

I'm happy to submit other non code style type fixes for the time being as I'm going through the code to see what's what.

Like I say, a longer term goal of having things enforced by CI, and also automatically fixable too (`phpcbf` using phpcs rules)... Spaces and braces generally increase readability, but things like position of braces is YMMV and personal opinion, so are a bit more contentious, but at least being consistent is a good thing

Any idea when you might be done with that with the fork merges?

TrystanLea, 2019-07-03:
Great thanks @reedy how does PHPCS work, do you run it as part of the github travis testing? If its just providing an advisory rather than applying styling rules globally then thats fine. I prefer doing it that way so that I can see the changes being suggested/applied in a gradual step by step way.

reedy, 2019-07-03:
It will be runnable locally too, and for some rules, it'll have an "autofix" functionality too. You can define a set a rules for a repo, and then set exclusions (or not set the rules to begin with) on files etc.

As the repo is relatively small, it shouldn't take too long to run over all files (but don't quote me on that just yet :))

I believe you can downgrade all errors just to warnings too, so you can see what's wrong, without travis going lolno

So, as you say, we can gradually improve things, enabling new rules, cleaning up so they don't fail, and then preventing regression in the future. There's some stuff that'd I'd suggest we/you do earlier in the process (whitespace for example) and then look at rules for spacing inside functions, conditions and loops etc at a later date. Basically with the end goal of having a "style guide" for people to work towards/against

borpin, 2019-07-03:
Could try this on some of the smaller Module repos first.

reedy, 2019-07-03:
> Could try this on some of the smaller Module repos first.

Yeah, it's probably not a bad idea, especially if we're wanting to avoid big sweeping changes here for a little while.

I've already done some manual cleanup over at https://github.com/emoncms/wifi and have made a start creating the travis config and composer doing basic php linting (to make sure it's valid code) over that and most of the other active/non-abandoned PHP repos in the emoncms org

Then we also have a target phpcs config to merge into here a little bit further down the road

MyForest, 2020-02-26:
Just an FYI.

In my teams I work with elsewhere we put the style rules as parts of the Docker build and we fail the build if the style rules fail. That means people submitting PRs get notified there's a problem and get a chance to fix that before the owner comes along to review the PR. Of course some people get frustrated that submitting a PR is hard because of this so it might not suit Emoncms.

We typically use things like [black](https://black.readthedocs.io/en/stable/), [prettier](https://prettier.io/) and [pycodestyle](https://pypi.org/project/pycodestyle/) and try very hard to not tinker with the rules.

reedy, 2020-02-26:
I don't think it's unreasonable to enforce code style as part of the CI/submission process. If it requires a human to manually review code style, you're doing it wrong. And in most cases, you should be able to automate the style fixing too...

It's pretty common for open source projects to have these sorts of requirements these days. Just because it's "free" doesn't mean we should hold ourselves to a lower standard.

To some extent, having standardised code style makes it easer to contribute. You have something to follow.

However, it shouldn't fail for unrelated code. So... Either you only apply the rules to code changed (harder), or you at least fix the whole repo first. Depending on how you do it, incrementally turning on rules, or one big patch fixing all the things.

Certainly being able to test locally is a benefit, along with automated fixing. So you can have it as commit hooks...

reedy, 2020-04-24:
The trailing whitespaces, inconsistent indenting.. among others are really starting to irk me

TrystanLea, 2020-04-25:
I need to really understand what we are proposing here to progress on this.  So I see PHP_CodeSniffer here: https://github.com/squizlabs/PHP_CodeSniffer and install it with 

    pear install PHP_CodeSniffer

I then test the default coding standard against emoncms/route.php I get:

```
/var/www/emoncms$ phpcs route.php 

FILE: /var/www/emoncms/route.php
--------------------------------------------------------------------------------
FOUND 25 ERRORS AND 7 WARNINGS AFFECTING 31 LINES
--------------------------------------------------------------------------------
   2 | ERROR   | [ ] You must use "/**" style comments for a file comment
  18 | ERROR   | [ ] Missing doc comment for class Route
  20 | ERROR   | [ ] Missing short description in doc comment
  25 | ERROR   | [ ] Missing short description in doc comment
  30 | ERROR   | [ ] Missing short description in doc comment
  35 | ERROR   | [ ] Missing short description in doc comment
  40 | ERROR   | [ ] Missing short description in doc comment
  45 | ERROR   | [ ] Missing short description in doc comment
  50 | ERROR   | [ ] Missing short description in doc comment
  55 | ERROR   | [ ] Missing short description in doc comment
  60 | ERROR   | [ ] Missing short description in doc comment
  61 | ERROR   | [ ] Missing parameter comment
  62 | ERROR   | [ ] Missing parameter comment
  63 | ERROR   | [ ] Missing parameter comment
  69 | WARNING | [ ] Line exceeds 85 characters; contains 136 characters
  72 | ERROR   | [ ] Missing short description in doc comment
  73 | ERROR   | [ ] Missing parameter comment
  73 | ERROR   | [x] Tag value for @param tag indented incorrectly; expected 1
     |         |     spaces but found 2
  74 | ERROR   | [ ] Missing parameter comment
  75 | ERROR   | [ ] Missing parameter comment
  76 | ERROR   | [ ] Missing @return tag in function comment
  81 | WARNING | [ ] Line exceeds 85 characters; contains 124 characters
  82 | WARNING | [ ] Line exceeds 85 characters; contains 124 characters
  89 | WARNING | [ ] Line exceeds 85 characters; contains 128 characters
  92 | WARNING | [ ] Line exceeds 85 characters; contains 97 characters
  95 | WARNING | [ ] Line exceeds 85 characters; contains 97 characters
  97 | WARNING | [ ] Line exceeds 85 characters; contains 124 characters
 134 | ERROR   | [ ] Expected "if (...) {\n"; found "if(...) {\n"
 136 | ERROR   | [ ] Expected "} elseif (...) {\n"; found "} elseif(...) {\n"
 138 | ERROR   | [ ] Expected "} elseif (...) {\n"; found "} elseif(...) {\n"
 140 | ERROR   | [ ] Expected "} elseif (...) {\n"; found "} elseif(...) {\n"
 149 | ERROR   | [ ] Missing short description in doc comment
--------------------------------------------------------------------------------
PHPCBF CAN FIX THE 1 MARKED SNIFF VIOLATIONS AUTOMATICALLY
--------------------------------------------------------------------------------

Time: 152ms; Memory: 6MB
```
I'd be happy to apply the spacing recommendations 134-140 that's how I generally do it when I code.

Perhaps we drop the "missing short description in doc comment" recommendation and the line 2 error, these seem a bit arbitrary, the comments in this file are pretty well formed Id say already. Id also like to drop the 85 character limit on a line.

TrystanLea, 2020-04-25:
Here's the result with PSR2 standard:

```
/var/www/emoncms$ phpcs --standard=PSR2 route.php 

FILE: /var/www/emoncms/route.php
-----------------------------------------------------------------------------------------------------
FOUND 5 ERRORS AND 6 WARNINGS AFFECTING 11 LINES
-----------------------------------------------------------------------------------------------------
   1 | WARNING | [ ] A file should declare new symbols (classes, functions, constants, etc.) and
     |         |     cause no other side effects, or it should execute logic with side effects, but
     |         |     should not do both. The first symbol is defined on line 18 and the first side
     |         |     effect is on line 16.
  18 | ERROR   | [ ] Each class must be in a namespace of at least one level (a top-level vendor
     |         |     name)
  69 | WARNING | [ ] Line exceeds 120 characters; contains 136 characters
  81 | WARNING | [ ] Line exceeds 120 characters; contains 124 characters
  82 | WARNING | [ ] Line exceeds 120 characters; contains 124 characters
  89 | WARNING | [ ] Line exceeds 120 characters; contains 128 characters
  97 | WARNING | [ ] Line exceeds 120 characters; contains 124 characters
 134 | ERROR   | [x] Expected 1 space(s) after IF keyword; 0 found
 136 | ERROR   | [x] Expected 1 space(s) after ELSEIF keyword; 0 found
 138 | ERROR   | [x] Expected 1 space(s) after ELSEIF keyword; 0 found
 140 | ERROR   | [x] Expected 1 space(s) after ELSEIF keyword; 0 found
-----------------------------------------------------------------------------------------------------
PHPCBF CAN FIX THE 4 MARKED SNIFF VIOLATIONS AUTOMATICALLY
-----------------------------------------------------------------------------------------------------

Time: 175ms; Memory: 6MB
```
That's better, agree again on the 134-140 white space requirements. 
Dont agree on the character limit, but its good to see it has increased.
Not sure about the first 2

TrystanLea, 2020-04-25:
Im happy with the result of running:

    phpcbf --standard=PSR2 route.php

Running the same on index.php, generally happy with the result, but things like:

```
- if (isset($_GET['q'])) $log->info($_GET['q']);
+if (isset($_GET['q'])) {
+    $log->info($_GET['q']);
+}
```

seem a bit unneeded to me.. or:

```
-    if (get('embed')==1) $embed = 1; else $embed = 0;
+    if (get('embed')==1) {
+        $embed = 1;
+    } else {
+        $embed = 0;
+    }
```

When they are that short, I think it makes sense to keep it in a condensed format..

chaveiro, 2020-04-25:
Hi Trystan, i understand you in the sense that many of us that are self taught programmers took shortcuts and simplifications throughout our lives that makes sense.
The other side is that best practices are also important and new developers learn to follow them. 
That is good, so that everybody expectations are the same and less mistakes arrise.
So usualy just adjust validation tools thrushold that team is comfortable with  and follow much standardized best practices as possible.

reedy, 2020-04-25:
I would disagree. Not using braces caused major bugs in things like OpenSSL over the years. It's easier (and better IMHO) to be explicit

https://blog.codecentric.de/en/2014/02/curly-braces/

The "experts" get it wrong, so it wouldn't surprise me if other people did too.

Basically, if you have a rule you don't want to fix you can add exclusions in a .phpcs.xml file, or on the command line

https://github.com/squizlabs/PHP_CodeSniffer/wiki/Advanced-Usage#limiting-results-to-specific-sniffs

Or we can try and build a repo with all the rules we want to apply/use and then use composer to have this installed on all repos for the linting

Some issues are definitely less important. Missing function doc? Yeah, it's not great, but we can fix it later etc. Especially as it's not an "automated" fix

Then other ones like Class per file is generally considered good practice, and while we're a way away from doing this, is helpful for doing things like PSR-4 autoloading

Again, if it's not things you/we want to fix now, we can exclude them; on a pattern/file/repo basis and come back to them later. Fixing incrementally is easier

reedy, 2020-04-25:
> I need to really understand what we are proposing here to progress on this. So I see PHP_CodeSniffer here: https://github.com/squizlabs/PHP_CodeSniffer and install it with
> 
> ```
> pear install PHP_CodeSniffer
> ```

Don't do that :). Use composer!

reedy, 2020-04-25:
Noting, there are many 3rd party rulesets available elsewhere, things can be configured etc

There's no right answers (well, beyond consistency for example, it should be tabs or space indenting for "in house" code, not both, and certainly not both in the same file. And if spaces, how many spaces for a level of indenting. All adds to readability).

TrystanLea, 2020-04-25:
Thanks both, any views on the different standards? I think Im happy with most of the PSR2 standard reading here looks like everything above that is just more detail on specific components? 

Maybe I start here by applying PSR2 to a couple of files to see where I get too and just accept some of the results Im less keen on..

reedy, 2020-04-25:
A slightly amusing one is [PSR-2](https://www.php-fig.org/psr/psr-2/) is deprecated in favour of [PSR-12](https://www.php-fig.org/psr/psr-12/)

Personally, I'm more in favour of more whitespace where possible, it generally adds to readability, and it doesn't make any difference unlike JS (which is why JS minification is a thing). But the PSR-2 rules seem to fix up my pet peeves, and at least make things consistent, which is more important for me than anything else

Many bigger projects take some elements of standardised style, but then add their own tweaks, stricter versions and stuff. It all depend son the amount of people involved, and whether people agree, disagree or don't really care

Obviously this isn't final, we can add more later, or tweak them etc. It's very much an ongoing effort :)

When we have something we like, I don't mind rolling it out to the other repos too

TrystanLea, 2020-04-25:
Thanks @reedy, not such a fan of all the new lines in PSR-12 e.g https://quickadminpanel.com/blog/psr-2-and-psr-12-why-we-need-standards-and-how-to-apply-them/  maybe we go for PSR-2 for now?

Id like to start by making these changes myself, I don’t like merging pull requests with tens of thousands of changes and it will give me a chance to evaluate the effect on a wider range of code in the project.

reedy, 2020-04-25:
I just made #1588 for fun... (if you want to use it for reference) There's probably gonna need to be some manual fixing after some of the changes

Oh, and admin_controller.php is seemingly not fixable automatically... :D

reedy, 2020-04-25:
I suspect, commiting a directory/handful of files at a time is a probably a better way to do it and make it easier to review

TrystanLea, 2020-04-29:
A small start: https://github.com/emoncms/emoncms/commit/81b2534b51c389766ada1ae63692098be999a8ec :)

TrystanLea, 2020-04-29:
and https://github.com/emoncms/emoncms/commit/d8a9b177f5c608c6d830cd4277497500aad373cd

how do you apply recursively to only .php files?

reedy, 2020-04-29:
> how do you apply recursively through multiple directory levels to only .php files?

As I did it in https://github.com/emoncms/emoncms/pull/1585 ;)

```
phpcbf --standard=PSR2 --ignore=*/vendor/* --ignore=*/Lib/* --extensions=php .
```

reedy, 2020-04-29:
I do note that the moving of the `{` from the end of a function defintion onto a new line seems pointless... Especially when they're staying on the end of the line in `if` statements...

Yay, consistency

TrystanLea, 2025-07-18:
Pinning this issue to the top as it's currently actively being worked on.

TrystanLea, 2025-07-18:
Thanks @reedy reading up very briefly on this, I guess our target should be full PSR-12? So steps in that direction ?