emoncms/emoncms #1341: changing window width brings left-side menu out

danbates2, 2019-07-02:
when changing window width the left hand side menu slides out.

TrystanLea, 2019-07-03:
Hey Dan, this is intended behaviour, when do you find it an issue?

chaveiro, 2019-07-05:
Generating an action (windows resize) that is not related in anything to a menu appearing seemed to me like  a bug.
If it's be design than it's a nuisance. 
You know what i mean if you use a 50" 4k monitor where you adjust window sizes regulary.

danbates2, 2019-07-09:
@TrystanLea bump :)

TrystanLea, 2019-07-11:
I dont agree that its a bug but one option could be to make a distinction that when the user clicks on the sidebar top level icon, which hides the sidebar that it then does not reopen when the window is expanded?

danbates2, 2019-07-11:
That'd be great!

TrystanLea, 2019-07-11:
Linking this issue to https://github.com/emoncms/emoncms/issues/1348 as very similar

TrystanLea, 2019-09-03:
This is now fixed in latest version

closed 2019-09-03 by TrystanLea