emoncms/emoncms #1346: Development: hierarchical settings

TrystanLea, 2019-07-03:
Continuing on our discussion here: https://github.com/emoncms/emoncms/pull/1010

and noting your comment here @borpin https://github.com/emoncms/emoncms/pull/1345 @emrysr and I have been discussing this this morning, we agree that there are nice advantages to the ini file, in particular being able to load the settings from python or other languages. We've merged in your settings-ini branch @takkaria into a development branch based of the latest emoncms here https://github.com/emoncms/emoncms/tree/settings-ini

We've also solved the array merge issue for the php approach. See: https://github.com/emoncms/emoncms/commit/35812a4d982894bd0f742a3f17af9d7c25ec467d

For now we are going to support both options, emoncms.org will likely use the php approach as the file load part is ~10x faster, although I might try testing both to see if it is an issue. I see the default moving to using the .ini approach on emonSD/standard installs.

We are going to implement a test of storing all settings in the single settings object and accessing these directly in emoncms (as parsed by @takkaria's ini parse) rather than re-mapping to all the different variables - that's the next step.

borpin, 2019-08-13:
Noting PR #1383, I have a couple of comments.

I still think that this is making things harder not easier for the developer side.  Each time a setting is added, it needs to be added to the `process_settings.php`.

It would be better (IMHO) if the settings file did not have sections so the variables can be created dynamically (see https://stackoverflow.com/a/10214530/5956071) from the key name.

The process_setting file would then only need to cater for existing arrays. All future settings are created as standalone items and automatically catered for.  In addition, where a setting is used just once or twice, it's probably better to change where it is used and remove the need for the array kludge.

Could the initial **writing** of a `settings.ini` file be done as part of the update process, and then move the `settings.php` to a `settings.old` file?

This writing will be much more difficult as ini sections have been used (which was partly why I didn't when I first created this).

Second thought, how about a setting to enable the `ENV_VARS` so this is deliberate rather than accidentally included.

TrystanLea, 2019-08-13:
Thanks @borpin my other thought is that we should change the way settings variables are used in emoncms itself, so that its always accessed through a single settings array, rather than having multiple different global variables?

The main question, is how to then support backwards compatibility to the existing flat php settings format.. perhaps it would be ok to make it a breaking change?

TrystanLea, 2019-08-13:
E.g refactor the php settings version to the format:

    <?php
    $settings = array(

        //1 #### Mysql database settings
        "database"=>array(
            "server"=>"localhost",
            "database"=>"emoncms",
            "username"=>"_DB_USER_",
            "password"=>"_DB_PASSWORD_",
            "port"=>3306,
             // Skip database setup test - set to false once database has been setup.
            "dbtest"=>true
        ),
        
        //2 #### Redis
        "redis"=>array(
            "enabled"=>false
            'host'=>'localhost',
            'port'=>6379,
            'auth'=>'',
            'dbnum'=>'',
            'prefix'=>'emoncms'
        ),

and then in index.php:

    $mysqli = @new mysqli(
        $database["server"],
        $database["username"],
        $database["password"],
        $database["database"],
        $database["port"]
    );

borpin, 2019-08-13:
Can I just check my understanding, the idea is to (eventually) use a settings object, created directly from a `parse-ini` directly every time a value is required within the whole code?

If so, then yes, this is the way to go.

Rather than refactor, I'd do it as part of an upgrade; read the current settings file and write a new `settings.ini`, then move the `settings.php` to `settings.old`. Job done.

You can just use the existing process-settings process for now (although it will now process just 2 ini files hierarchically) and gradually replace the individual variables with direct access of the settings object and then remove that variable from the process_settings file as you go.

borpin, 2019-08-13:
Next task will be allowing the editing of the settings file from the web interface 😀 (I'll get my hat...)

TrystanLea, 2019-08-13:
I've made a start on this idea, got it working as far as logging in, and viewing a feed https://github.com/emoncms/emoncms/compare/settings-ini-v2 

There are a lot more settings to update to get it all working, but the concept is there. 
What do you all think?

borpin, 2019-08-13:
I really do think it is a waste of time updating the php settings file. Just convert the settings to an ini file on upgrade, then work on using the `$settings` object directly throughout.

chaveiro, 2019-08-13:
Is a new config format really important / needed ? 
I could point out other options to INI like XML, TOML, YAML, HJSON,  you name it.
See this article, and what they say about sticking with own scripting language (like we always hadn in emoncms):
https://www.lucidchart.com/techblog/2018/07/16/why-json-isnt-a-good-configuration-language/

Accessing emoncms 'settings' from other languages can be archived via a new API for the purpose, and it's a better architecture anyway.
But in the end would stick with if it's not broke don't fix it mantra...

borpin, 2019-08-14:
> But in the end would stick with if it's not broke don't fix it mantra...

Thing is it is broken. Adding in new settings is a PITA with users often required to completely rebuild their settings file or accept the default.

The inbuilt PHP ability to parse the ini file is the main plus to using that format.

This change will have minimal impact on the majority of users but should make it easier for those that do not want the standard settings.  In addition, it will make the introduction of new settings easier.

pb66, 2019-08-14:
We should try not to confuse how it is coded with how emoncms is used.

I agree we need to do things differently, but I don't think that necessarily means re-writing huge chunks of emoncms. 

The way that process_settings.php works currently almost negates any need for a settings file. Most if not all settings are met with "if $xyz is not set, then $xyz = some default" type code, this is in fact the real default settings and the "default.settings.php" files are actually just templates for a "settings.php" file that probably isn't even needed on a vanilla install (if it wasn't insisted upon).

Why not have process_settings.php use default.settings.php directly as a proper source of default settings configured in such a way that emoncms will fully function (on localhost) without a settings.php file at all. each setting is then set with "if settings.php $xyz is not set then $xyz = default.settings.php $xyz" type code. Then only local user changes need to be put in settings.php and that file will never get touched by an emoncms update or git pull etc.

As and when new settings get added to the process_settings.php they are also added to the default.settings.php with the recommended setting as per usual and when emoncms is updated the revised setting is pulled in and set to the default without touching settings.php, if the user then want to change the default behaviour, they then edit settings.php to add just that one setting.

Using it this way rather than the way we are currently using it fixes most issues without major code changes or issues with backwards comparability (except for changing the default behaviour for settings already explicitly set in a users existing settings.php).

I have never really been a fan of the current format of the settings.php file, but that is a different thing to how we use what we have.

I am interested in what's being discussed about hierarchical settings, how setting are handled in emoncms and different file formats, but I am far more concerned that an overly complex solution to a relatively straightforward problem might be on the horizon. That's not to say those changes aren't wanted, just that they are not needed to fix the immediate problem of updating the settings.php during a git pull, there are simpler solutions on hand.

With time available for development always in short supply, there are perhaps area's that are more "broken" than this might appear.

A simple config editor like the one used for emonHub via emoncms would (IMO) have a far greater and lasting impact, the settings that are "irretrievably breaking" changes can be explicitly excluded (eg mysql server settings) from editing (ie read-only for viewing) the save button could do a check and only save if the settings file content is valid code (and doesn't change mysql settings) after user editing.

borpin, 2019-08-14:
> As and when new settings get added to the process_settings.php they are also added to the default.settings.php with the recommended setting as per usual and when emoncms is updated the revised setting is pulled in and set to the default without touching settings.php, if the user then wants to change the default behaviour, they then edit settings.php to add just that one setting.

The problem with this is 
1. You need to amend 2 files to introduce a setting.
2. If the new setting is a breaking change (like the mqtt settings array) the user **has** to go through the whole settings file, work out what has changed then transpose all their personal settings one way or the other while having a broken system.

Moving to a hierarchical system will mean any setting can be introduced to the default file and then used directly. The concept is not particularly complex, read the default, then the local file and overwrite the default setting with a local setting (if one exists - local file could be empty).

There is no *need* to rewrite anything else for this to work (except the process setting file to get the values from the settings object which has been done); using the setting object directly can be introduced over time.

There should be no backwards compatibility issues.

The other advantage is the ability to easily introduce ENV vars rather than have to hardcode them.

TrystanLea, 2019-08-24:
Ok Im pretty much there now with this new settings system, the latest version is available here: https://github.com/emoncms/emoncms/tree/settings-ini-v2

There are two settings format: php and ini and Im keen to keep them both, php is a bit faster - given that it will be pre-compiled/cached by the php system. 

Here's the default settings object in php:
https://github.com/emoncms/emoncms/blob/settings-ini-v2/default-settings.php

and the same in the ini format (looks really neat):
https://github.com/emoncms/emoncms/blob/settings-ini-v2/default-settings.ini

The main question I have at this point is: What should the user created settings file look like? Given that the settings file does not have to contain all settings what settings should it contain? 
as a starting point I have created an example settings file in both the php and ini format that contain the emonpi specific settings

emonpi php example:
https://github.com/emoncms/emoncms/blob/settings-ini-v2/emonpi.settings.php

emonpi ini example:
https://github.com/emoncms/emoncms/blob/settings-ini-v2/emonpi.settings.ini

What do you think @borpin ?

borpin, 2019-08-25:
> There are two settings format: php and ini and Im keen to keep them both

I think this is a mistake and is going to lead to confusion in the future.  If they are different (PHP & INI), which takes precedence?

> php is a bit faster

Does the system read and parse it every time? How about a reload settings facility under admin?

> What should the user created settings file look like?

Going forward, I think the default files should work if the scripts are used to install the system. One point about the script install was to get away from the pi/emonpi centric installs.

For now it should just be what is needed for an emonpi based system to work as it did before.

I still think there needs to be a migration script that takes the current (in use) PHP settings and writes them to an INI file that is used in the future.

It could remove the need for the Process-settings file to assess individual settings (except to import the settings).

But YMMV.

TrystanLea, 2019-08-25:
> For now it should just be what is needed for an emonpi based system to work as it did before.

Would you be happy with: 
https://github.com/emoncms/emoncms/blob/settings-ini-v2/emonpi.settings.php
or should it include a few more settings?

We can automate migration for emonSD installs that have not modified the settings file, that would be really easy, but beyond that I think it gets quite complex and its not something I'm going to get time to work on. We have several times in the past required that the user recreates the settings file after introduction of new settings so I don’t think this last time is a big issue.

borpin, 2019-08-25:
Just needs the bare minimum to take the default to what the emonpi needs to be backward compatible.

> If they are different (PHP & INI), which takes precedence?

What is the answer to this?

Still think it is a bad idea to have both...

TrystanLea, 2019-08-25:
So far I've been testing with the php version, but I do like the look of the ini format and would be happy to standardise on the use of the ini format for emonSD

TrystanLea, 2019-09-03:
This is what the cut down ini settings file looks like for the non standard emonpi settings:

    [sql]
    username = "emoncms"
    password = "emonpiemoncmsmysql2016"

    [redis]
    enabled = true
    prefix = ''

    [mqtt]
    enabled = true
    user = 'emonpi'
    password = 'emonpimqtt2016'

    [feed]
    engines_hidden = [0,6,10]
    redisbuffer[enabled] = true
    phpfina[datadir] = 'var/opt/emoncms/phpfina/'
    phptimeseries[datadir] = '/var/opt/emoncms/phptimeseries/'

    [interface]
    enable_admin_ui = true
    feedviewpath = "graph/"

An alternative could be to include a few more settings so as to give a more complete appearance: 

    [sql]
    server   = "localhost"
    database = "emoncms"
    username = "emoncms"
    password = "emonpiemoncmsmysql2016"
    port     = 3306

    [redis]
    enabled = true
    host = 'localhost'
    port = 6379
    auth = ''
    dbnum = ''
    prefix = ''

    [mqtt]
    enabled = true
    host = 'localhost'
    port = 1883
    user = 'emonpi'
    password = 'emonpimqtt2016'
    basetopic = 'emon'
    client_id = 'emoncms'

    [feed]
    engines_hidden = [0,6,10]
    redisbuffer[enabled] = true
    redisbuffer[sleep] = 600
    phpfina[datadir] = 'var/opt/emoncms/phpfina/'
    phptimeseries[datadir] = '/var/opt/emoncms/phptimeseries/'

    [interface]
    enable_admin_ui = true
    enable_update_ui = true
    enable_multi_user = false
    feedviewpath = "graph/"
    default_language = 'en_GB'
    themecolor = "blue"

TrystanLea, 2019-09-23:
The hierarchical settings branch is now merged with master. I made it backwards compatible to use the old settings file if present for now.

The old settings file is translated into the new settings object here:
https://github.com/emoncms/emoncms/blob/master/Lib/process_old_settings.php
this is only called if the $settings variable is not present in the settings.php file.

I've created an example.settings.ini file here: 
https://github.com/emoncms/emoncms/blob/master/example.settings.ini
and updated the installation docs to reflect configuration using the ini file format (php settings is still possible for those that prefer).

I will be updating EmonScripts next to use the new ini format.
I will

TrystanLea, 2019-09-23:
I realised that I was getting quite stuck with this development as its a significant change. By introducing the backwards compatibility - the ability to parse in the old settings format its possible to separate out the settings object code change from the process of updating all systems to use a new settings file format. The merge so far is the first step.

TrystanLea, 2019-09-23:
The only issue I can see is with systems using the environment variables, those using this configuration need to copy settings.env.ini to settings.ini https://github.com/emoncms/emoncms/blob/master/settings.env.ini

borpin, 2019-09-24:
@TrystanLea - great news - I'd suggest making an announcement and asking folk to pull the master so it can get a bit of testing, then push a 10.X.0 update before making any other significant changes. to the Repo.

Could I also make an observation that it would be good to push a 'patch' before merging major changes - just helps with rolling back if necessary.

> this is only called if the $settings variable is not present in the settings.php file.

I'm not sure I understand this - is this a new setting in `settings.php`?

So under normal circumstances, **IF** there is no `settings.ini` found **or** no ` $settings` variable in settings.php, then `process_old_settings.php` is called and the old `settings.php` file is processed.

Is that right?

TrystanLea, 2019-09-24:
Done: https://community.openenergymonitor.org/t/new-emoncms-settings-format-help-testing-needed/12002 

> So under normal circumstances, IF there is no settings.ini found or no $settings variable in settings.php, then process_old_settings.php is called and the old settings.php file is processed.

Yes that is correct. The check for the $settings variable is in case the user is using the new $settings object php settings file format. If that's not found it assumes the settings file is the old flat variable format.

closed 2019-11-25 by TrystanLea