emoncms/emoncms #1349: Feeds table lost information

chaveiro, 2019-07-05:
Important details used for adminitration of feeds disapeared on V10.
If the less usefull elements disapearred when the screen size is small i would understand but that does not seem the case.

See the image bellow. 
Left side is v10, right side  v9.

The problems are signaled by the arrows.
1 - Yellow: Here did the virtual feeds processlist  go ?
2 - How the check datatype?
3 - How to bulk export an entire feed TAG ?

[image]

TrystanLea, 2019-07-08:
Noted, the virtual feed processlist and bulk export was accidentally missed out, I don’t think the datatype is that useful, but we could add it into the hover over tooltip.

borpin, 2019-07-09:
How about the trim function that I think was accessed via the cog?

TrystanLea, 2019-07-11:
Sorry, the bulk export is present, you need to click on the tick box selectors on the left hand side to select the feeds that you wish to export. This also works for deleting and timing multiple feeds.

The process list is not shown, but creating and adding processes to virtual feeds does work.

closed 2019-07-19 by TrystanLea

chaveiro, 2019-07-19:
Still on the topic of bulk export, how difficult is to add a quick way to select/deselect a full tag of feeds?
Remember it was a request by my time of developing emoncms.
Say you want to export manually some devices and each device have 15 feeds. 
On v9 it's a 2 clicks step, on v10 it is a 16 clicks step.
And to export another device, a 31 clicks step to deselect previous and select the next device...

TrystanLea, 2019-07-19:
Hello @chaveiro, @emrysr has been working on a feature that allows for this 2 click approach for the inputs interface, currently not merged, we will work towards implementing the same in the feeds list https://github.com/emoncms/emoncms/pull/1360

PR #1369 (2019-07-19): added datatype to feedlist tooltips
fix #1349 

[image]