emoncms/emoncms #1350: BUGs on v10 Dashboard

chaveiro, 2019-07-06:
Something core got really messed up on v10 upgrade.
See bellow screnshots of the same dashboard.
Left is v9, right is V10, using the exact same database with db upgrade done.

See arrows:
BUG1: yellow: **Multigraph** daily feed (Kwhd) no longer starts at 00:00 but is starting at 12:00pm ?
BUG2: red: **Zoom** graph does not allow to select current month / year when you zoom out.
BUG3: red large: **Histogram** does not show anything.

[image]

TrystanLea, 2019-07-08:
Noted these @chaveiro to look into

TrystanLea, 2019-07-11:
Hello @chaveiro Im unsure what changes in v10 would have caused this. Could you help us isolate the exact release version that results in these issues. Could you try version 9.9.9 to start with? and then perhaps 10.0.0?

TrystanLea, 2019-07-11:
The Zoom visualisation appears to work for me, for current month and day but the year view is showing as blank. I don’t have any daily or histogram data to test with unfortunately..

borpin, 2019-07-11:
> Could you try version 9.9.9 to start with?

I tried this and got fatal php errors - too many changes to other related modules I suspect.

TrystanLea, 2019-07-12:
Ok, Are you able to replicate the issues @chaveiro is seeing in v10 @borpin?

borpin, 2019-07-12:
No. I tried to go back and see when the other issue (cog gone from feeds) occurred but could not go back to before 10.0.0. I guess incompatibility with other modules.

TrystanLea, 2019-07-12:
Ah great, the cog disappearing was by design - my bad, its replaced by the cog that appears at the top of the page after multi feed selection.

borpin, 2019-07-12:
But a cog does not appear? Oh I see to trim a feed you need to select 'delete'. Not very intuitive!!! Can the tool tip be changed to 'delete / trim' please?

pb66, 2019-07-12:
> Oh I see to trim a feed you need to select 'delete'. Not very intuitive!!!

I agree, users generally won't click a delete button unless the're intending to delete the whole feed (or at least don't mind if that is what transpires), there is no way to know in advance if there will be any confirmation before deletion or your data will just vanish!

I spent a fair bit of time specifically looking for the trim feature for this issue on the forum

https://community.openenergymonitor.org/t/wired-as-type-1-but-will-set-up-only-as-type-2/10979/19?u=pb66

although this example was on an older emoncms version, the position is the same.

A tooltip could be worded to reassure users that clicking "delete" at this stage isn't terminal and further options/features will be presented.

borpin, 2019-07-12:
It is also far too easy to have multiple feeds selected without realising it!!!!!

This needs some serious confirmatory messages before it does it, including information about what feeds are selected etc.

chaveiro, 2019-07-15:
Just to be clear, the 9 version i refer here to be using is 9.8.10 | 2017.08.17 with a dashboard version from that time. I dind't upgrade for a long time. 
Anyway i've made a test, use the dashboard from that time on top of lastest v10 core and the problem is the same, meaning it's not related to the dashboard but to the data that is feeding it.

borpin, 2019-07-15:
@chaveiro, have you tried recreating the graphs from scratch? This will identify if it is a problem with the graph itself or the migration process.

chaveiro, 2019-07-17:
I tried just now and got this error creating a zoom graph.
Was this tested before?
[image]

TrystanLea, 2019-07-17:
Its working fine for me here, yes I did a lot of testing before releasing the "feed selection mode" feature. There was a forum thread and testing branch for some time if I recall. Its a long time back now, before v10.

borpin, 2019-07-17:
Have you done a database update? I am suspecting so. Are you sure all the repos did actually update (failures tend to be silent).

I'm not seeing this in either of my installs (one on master and one on stable)
[image]

TrystanLea, 2019-07-17:
Here's the forum thread with links to associated github issues 
https://community.openenergymonitor.org/t/portable-emoncms-dashboards-looking-for-testers/6870

borpin, 2019-07-17:
What version does the admin page say the dashboard module is at @chaveiro? Should be 2.0.3

For me, `git log` looks like this

```
pi@emonpi:/var/www/emoncms/Modules/dashboard $ git log -4
commit 337df804a94e9225de2d702d94477f9926d4154c
Merge: abd1a0b 43c10ae
Author: Glyn Hudson <glynhudson@users.noreply.github.com>
Date:   Thu Jun 20 15:05:13 2019 +0100

    Merge pull request #222 from emoncms/master

    V2.0.3

commit 43c10ae920a30a75c93d811ef48ae83ee28a2e7a
Author: Glyn Hudson <glynhudson@users.noreply.github.com>
Date:   Thu Jun 20 15:03:30 2019 +0100

    V2.0.3 bump

commit 08d630c154b32336f42e9fd6549f7e215e61f04a
Merge: 8afa1e5 9e7880e
Author: Glyn Hudson <glynhudson@users.noreply.github.com>
Date:   Thu Jun 20 15:01:12 2019 +0100

    Merge pull request #220 from emrysr/feature/issue-82-public-alias

    Feature/issue 82 public alias

commit 8afa1e53412a49bde913effdd1fe84f1f97f32cc
Merge: 4e29f21 60961a2
Author: Glyn Hudson <glynhudson@users.noreply.github.com>
Date:   Thu Jun 20 15:00:36 2019 +0100

    Merge pull request #221 from emrysr/feature/hide-notification-once-added

    hide warning after addNew()
```

TrystanLea, 2019-07-17:
The error seems to be related to this line: https://github.com/emoncms/dashboard/blob/master/Views/dashboard_config.php#L35 which means $dashboard['feedmode'] is undefined. 

The dashboard is loaded from the model for use in the config interface here: https://github.com/emoncms/dashboard/blob/master/dashboard_controller.php#L63 and https://github.com/emoncms/dashboard/blob/master/dashboard_model.php#L185

Could you try saving the dashboard first @chaveiro and then reloading and checking the config?

chaveiro, 2019-07-17:
> Have you done a database update? I am suspecting so. 

I did but running it again and it showed something to be updated so i run and the dashboard editor now is working fine. :)

I'm using this:
[image]

borpin, 2019-07-18:
Excellent. You might need the graph module (separate repo) as well.

closed 2019-09-03 by TrystanLea