emoncms/emoncms #1351: emoncms_mqtt service no longer works on non pi systems

borpin, 2019-07-08:
This line breaks no pi systems.

When did we agree to this?????

https://github.com/emoncms/emoncms/blob/874f85c3eb8cf4aea563535d4df70992a05e5db0/scripts/services/emoncms_mqtt/emoncms_mqtt.service#L55

borpin, 2019-07-08:
The solution is fantastically simple.

During the install script create a file at `/etc/tmpfiles.d/emonpi.conf` with the contents
```
f /var/log/emoncms/${LOG_PATH} 0666 pi pi
```
where the required user and group could be collected from either the parent or the specified user in the script.

I think parent folders will be created by default.

Final comment, I know we discussed where to install these and settled on `/etc/` **however** it means I cannot override the file by creating one in `/etc/` so right now any git updates are screwed and would overwrite my changes anyway.

TrystanLea, 2019-07-08:
Thanks @borpin is that the only line needed in emonpi.conf. Can we create the parent folder if it does not exist? Im not sure that I understand your comment regarding /etc?

borpin, 2019-07-08:
I'm testing hereon my EmonPi, but something else is creating the folder and file as well. Contents of config file I'm using right now
```
d /var/log/emoncms - pi pi -
f /var/log/emoncms/emoncms.log 0666 pi pi -
```
For some reason I am getting an error if I try and set the directory permissions

My comment about `/etc` for the service files. In this case, If we had put the file in `/lib`, I could simply drop in a replacement service file into `/etc` and I would not impact the main files. As it is, because it is in `/etc` I have to edit the main file which will prevent a `git pull` as I have local changes.

I also cannot fix this with a 'drop-in' as I cannot override those commands.

borpin, 2019-07-08:
This works (I tried it with a non emoncms path) but something else is currently creating the folder and file.
```
d /var/log/emoncms 0755 pi pi -
f /var/log/emoncms/emonpi.log 0666 pi pi -
```
It also breaks the ENV variable thoughts as well!

Hey ho.

borpin, 2019-07-08:
So a little more experiment. By including these entries almost all of the `rc.local` can be removed
```
d /var/log/emoncms 0755 pi pi -
f /var/log/emoncms/emoncms.log 0666 pi pi -
d /var/log/emonpilcd 0755 pi root
d /var/log/emonhub 0775 root emonhub
d /var/log/supervisor 0755 root root
d /var/log/mosquitto 0755 mosquitto mosquitto
d /var/log/logrotate 0755 pi pi
d /var/log/openhab2 0755 root root
d /var/log/mysql 0755 mysql adm
d /var/log/apache2 0755 root adm
d /var/log/redis 0755 redis redis
```
There are some other entries in the `rc.local` that I suspect might be required for backwards compatibility.

I did need a `sudo systemctl daemon-reload` after changing the config file.

I suggest splitting these entries out to ones with a `pi` user and the rest.

I also suggest that as part of the install script we check on the use of `/var/log` on tmpfs (as `rc.local` does) and only create these config files if it is found to be there.

`rc.local` now contains
```
#!/bin/sh
#
# rc.local

# rc.local for Raspbian Stretch
echo" My IP address is":
hostname -I

# Create log dirs and files on tempfs partition
# Will only run if /var/log is mounted in tmpfs
if ( mount | grep "on /var/log "| grep -q "^tmpfs " )
then
  touch /home/pi/data/emonpiupdate.log
  ln -s /home/pi/data/emonpiupdate.log /var/log/emoncms/emonpiupdate.log

  service emonPiLCD restart

fi

# Run emonPi Update of first factory boot as Pi user (run condition > web connection exisits && ~/data/emonpiu$
su pi -c '/home/pi/emonpi/./firstbootupdate'

## Start Wifi AP see emonpi/wifiAP/readme.md
/home/pi/emonpi/wifiAP/startAP.sh
```

Of course Log2Ram would change this again 😄.

TrystanLea, 2019-08-05:
Thanks @borpin, I will get back to this, thanks for looking into it

borpin, 2019-09-09:
Grrr been caught out again by this stupid addition to the service files. Why make it so Pi specific.......

borpin, 2020-03-28:
This looks like it was solved at some point in time - can't remember when and can be closed @TrystanLea

borpin, 2020-05-20:
Reasonably sure this was fixed by #1410

closed 2020-05-20 by borpin