emoncms/emoncms #1361: Histogram Engine error

chaveiro, 2019-07-15:
After upgrading from v9 to v10 got this error returned while invoking: /feed/histogram.json?id=79&start=0&end=0&res=1

<br />
<b>Fatal error</b>:  Class Histogram contains 5 abstract methods and must therefore be declared abstract or implement the remaining methods (engine_methods::delete, engine_methods::get_feed_size, engine_methods::post, ...) in <b>/home/gtdev/sites/emondev.gtronica.com/Modules/feed/engine/Histogram.php</b> on line <b>5</b><br />

borpin, 2019-07-15:
I think it is down to the implementation of the shared_helper.  The problem is the Histogram class does not implement all functions.

By adding these it get past that point but as I do not have valid data to use, it falls over.
```
    public function delete($feedid){}
    public function post($feedid,$time,$value,$arg=null){}
    public function get_feed_size($feedid){}
    public function update($feedid,$time,$value){}
    public function get_data($feedid,$start,$end,$interval,$skipmissing,$limitinterval){}
```
It may solve your problem in the short run.

@emrysr seems to have done the changes to implement the shared_helper mechanism with #910 almost a year ago!

chaveiro, 2019-07-17:
Seems very few people upgraded and use this feature.
It's deceptioning that this kind of errors exists at this phase as it was cleary not tested.

TrystanLea, 2019-07-17:
Hello @chaveiro another possibility is that very few people use the histogram feature? I tried to discontinue it a number of years ago, but you objected :) I have stopped using it myself. I would prefer to discontinue it rather than trying to support a lot of different features, but Im also happy if you would like it kept in, would you be able to add the above functions to the shared helper and test with your histograms? that would be really helpful

chaveiro, 2019-07-17:
I confirm, adding that lines makes the engine work again.

borpin, 2019-07-18:
@TrystanLea perhaps those could be tidied (@emrysr)  up so if someone was to attempt to use them they do not simply fail quietly?  Fairly minor change.

TrystanLea, 2019-07-18:
Yes great that would be good, @chaveiro

TrystanLea, 2019-07-18:
Thanks @borpin, I've merged your suggested lines into the histogram engine https://github.com/emoncms/emoncms/commit/058300cae003c445fc4f2f8a200fe43e5aebc5be

closed 2019-07-18 by TrystanLea

borpin, 2019-07-18:
You could have at least added in a return message!!!!