emoncms/emoncms #1371: My Solar not storing data when wifi disabled

relft, 2019-07-21:
Hi,

I have detected that during night when WIFI is disabled My Solar is not storing data on My solar graph, but data on graphs are stored all night.

Why is not stored in this situation when all requirement are on this raspberry without needed any external resource?
Any suggestion how to solve this?

Thanks

TrystanLea, 2019-08-05:
Hello @relft how are you viewing the My Solar dashboard, if its on a seperate device to your emoncms installation e.g tablet --> emonpi and you disable the wifi it will not automatically pull in the data for the night time period. If you refresh the page with wifi enabled does the data reappear?

relft, 2019-08-05:
Query is made from a tablet to a raspberry pi, raspberry pi has wifi and
during night if we disablen wifi then no data is stored to raspberry pi,
data is captures from internsl card, no network requirement

Any suggestion?

Thanks

TrystanLea, 2019-09-03:
Hello @relft I dont quite understand Im afraid, you are welcome to post in full in your preferred language and then I can try and use google translate.

relft, 2019-09-08:
Hi,

Sorry, I'll try to give details

I have a raspberry with pican to get values from hpsu using emoncms, and if
during night we disable wifi nomdata is captured, and data is captured from
pican, not from lan or wifi

https://lamiacasaelettrica.com/rotex-hpsu-compact-hack-prima-parte/

Any suggestion

TrystanLea, 2019-12-12:
Hello @relft , sorry for the delay, is the data recorded in the emoncms feed? what happens when you reload the My Solar page?

relft, 2019-12-13:
Hi,

Finally, skme weeks ago migrated from wifi to cable lan and with this
works, not sure why with wifi fails

Thanks

closed 2020-01-20 by TrystanLea