emoncms/emoncms #1373: Fresh install 10.1.2, error when adding "new device"

teslabn, 2019-07-22:
Hi,
I installed emoncms on clean raspberry pi 3.
Now I have error message when I want to add new device.

EmonCMS Error
-----------------
Message: Uncaught ReferenceError: device_dialog is not defined
Route: Modules/input/Views/input_view.js?v=1
Line: 322
Column: 42

It will be good for you as solve bug, and for me to go step forward :)
Project is fantastic!
Best regards!

teslabn, 2019-07-22:
[image]
If you need more info about my instance, feel free to ask me. :)

MaffooClock, 2019-08-02:
I had the same problem.  After I installed the [Device module](https://github.com/emoncms/device), the error went away.

TrystanLea, 2019-08-05:
This issue should be fixed in the latest version, do you know which version and branch you are running?

teslabn, 2019-08-05:
@MaffooClock You are right.. When I installed Device module, I got full functionality. Thank you!
@TrystanLea Nice, my version is 10.1.2 (same on footer at web app, and admin view). Branch is master.

Thank you!

closed 2019-09-03 by TrystanLea