emoncms/emoncms #1380: graphing and virtual feed anomoly, problem with reset accumulators?

danbates2, 2019-08-10:
ref:
https://community.openenergymonitor.org/t/a-graph-or-virtual-feed-anomaly/11612/9

TrystanLea, 2020-01-21:
I can reproduce this issue by creating a virtual feed with a cumulative kWh feed as a source. There is a difference between the virtual and the original.

For example, selecting Graph  with interval: daily shows the following discrepancy between both feeds. Changing the interval to fixed interval and zooming in to hourly shows the same effect. This results in the kind of things @danbates2 is seeing.

[image]

@chaveiro I don’t understand enough about virtual feeds to fix this issue, it looks like a difference in timing between both approaches?

TrystanLea, 2020-01-21:
I can also reproduce @danbates2's 3 negative bars if the cumulative kWh feed resets:

[image]

Based on the following feeds:

[image]

TrystanLea, 2021-03-14:
Just a note to say that testing this again today this error is still present, there is a very slight difference between both feeds, it looks like a part of it is being caused by a time offset between the virtual and real feeds, @chaveiro I know you are busy, im not sure if this bug is worth looking into, it might well be being caused by a change since your 9.x version, or just compatibility with PHPFina.. Do you see any slight differences between one of your mysql source feeds and the virtual feed that clones the source feed?

closed 2025-07-16 by TrystanLea