emoncms/emoncms #1387: Cant create any input -> Spinner never ends

nixis, 2019-08-14:
Hi 👋,

I installed emoncms on a Debian 9.3 VM with the (new) installer-script and a 2nd time manually.
With both installations I am not able to create any inputs, because the spinner never stops.
I tried different browsers, deleted the cache and cookies.

[image]

The logfiles do not show any errors or similar, also the Chrome dev tools do not show any js errors or file load errors. But the list.json is loading every few seconds

[image]

<details><summary>Server Information</summary>

# Server Information
## Services
 
- **redis-server** :- **Active** Running

## Emoncms
 
- **Version** :- 10.1.4
- **Modules** :- Administration | App v2.0.2 | Dashboard v2.0.4 | Device v2.0.1 | EventProcesses | Feed | Graph v2.0.2 | Input | CoreProcess | Schedule | Time | User | Visualisation
- **Git** :- 
     - **URL** :- https://github.com/emoncms/emoncms.git
     - **Branch** :- * stable
     - **Describe** :- 10.1.4-1-ga82765de

## Server
 
- **OS** :- Linux 4.9.0-4-amd64
- **Host** :- emoncms | emoncms | (x.x.x.x)
- **Date** :- 2019-08-14 22:15:33 CEST
- **Uptime** :- 22:15:33 up  6:41,  1 user,  load average: 0.00, 0.00, 0.00

## Memory
 
- **RAM** :- Used: 16.39%
    - **Total** :- 1.96 GB
     - **Used** :- 328.52 MB
     - **Free** :- 1.64 GB
- **Swap** :- Used: 0.00%
    - **Total** :- 2 GB
     - **Used** :- 0 B
     - **Free** :- 2 GB

## Disk
 
- **/** :- Used: 5.58%
    - **Total** :- 29.4 GB
     - **Used** :- 1.64 GB
     - **Free** :- 26.25 GB

## HTTP
 
- **Server** :- Apache/2.4.25 (Debian) HTTP/1.1 CGI/1.1 80

## MySQL
 
- **Version** :- 5.5.5-10.1.38-MariaDB-0+deb9u1
- **Host** :- localhost:6379 (127.0.0.1)
- **Date** :- 2019-08-14 22:15:33 (UTC 02:00‌​)
- **Stats** :- Uptime: 5945  Threads: 1  Questions: 6828  Slow queries: 0  Opens: 27  Flush tables: 1  Open tables: 21  Queries per second avg: 1.148

## PHP
 
- **Version** :- 7.0.33-0+deb9u3 (Zend Version 3.0.0)
- **Modules** :- apache2handler | calendar v7.0.33-0+deb9u3 | Core v7.0.33-0+deb9u3 | ctype v7.0.33-0+deb9u3 | curl v7.0.33-0+deb9u3 | date v7.0.33-0+deb9u3 | dom v20031129 | exif v7.0.33-0+deb9u3 | fileinfo v1.0.5 | filter v7.0.33-0+deb9u3 | ftp v7.0.33-0+deb9u3 | gettext v7.0.33-0+deb9u3 | hash v1.0 | iconv v7.0.33-0+deb9u3 | json v1.4.0 | libxml v7.0.33-0+deb9u3 | mbstring v7.0.33-0+deb9u3 | mcrypt v7.0.33-0+deb9u3 | mysqli v7.0.33-0+deb9u3 | mysqlnd vmysqlnd 5.0.12-dev - 20150407 - $Id: b5c5906d452ec590732a93b051f3827e02749b83 $ | openssl v7.0.33-0+deb9u3 | pcre v7.0.33-0+deb9u3 | PDO v7.0.33-0+deb9u3 | pdo_mysql v7.0.33-0+deb9u3 | Phar v2.0.2 | posix v7.0.33-0+deb9u3 | readline v7.0.33-0+deb9u3 | Reflection v7.0.33-0+deb9u3 | session v7.0.33-0+deb9u3 | shmop v7.0.33-0+deb9u3 | SimpleXML v7.0.33-0+deb9u3 | sockets v7.0.33-0+deb9u3 | SPL v7.0.33-0+deb9u3 | standard v7.0.33-0+deb9u3 | sysvmsg v7.0.33-0+deb9u3 | sysvsem v7.0.33-0+deb9u3 | sysvshm v7.0.33-0+deb9u3 | tokenizer v7.0.33-0+deb9u3 | wddx v7.0.33-0+deb9u3 | xml v7.0.33-0+deb9u3 | xmlreader v7.0.33-0+deb9u3 | xmlwriter v7.0.33-0+deb9u3 | xsl v7.0.33-0+deb9u3 | Zend OPcache v7.0.33-0+deb9u3 | zlib v7.0.33-0+deb9u3
</details>

What can I do?

gitmeister, 2019-08-17:
Installed on my shared linux host (long time user) and I'm also getting the spinner on the INPUTS pane. If I switch to the Feeds or Admin pane the UI works fine. No error in emoncms log, no error in host log and I don't have any scripts being blocked by FF.

borpin, 2019-08-18:
@nixis @gitmeister Both these will be better discussed on the Community Forum https://community.openenergymonitor.org/latest

nixis, 2019-08-19:
@gitmeister https://community.openenergymonitor.org/t/cant-create-any-input-spinner-never-ends/11700

scfigg, 2019-08-21:
Same issue here with the pre-built RPi image on a RPi3 B+

nixis, 2019-08-21:
Solved - Inputs are automatically created.
Use MQTT: https://guide.openenergymonitor.org/technical/mqtt/
or HTTP(s) API: http://[YOUR-EMONCMS-IP]/input/api

closed 2019-08-21 by nixis