emoncms/emoncms #1390: Timeout Input page if no inputs discovered

borpin, 2019-08-21:
ref #1387
Suggest that the Inputs Page times out if no inputs have been found within a specific period of time or else, instead of the spinner, a message is displayed when no inputs have ever been found.

This is most likely to be useful to new installs on own hardware where no data has yet been specified.  The message could point to information about how to supply data to EmonCMS either by HTTP or MQTT.

chaveiro, 2019-08-21:
The spinner loading for ever is a bug, if no inputs are found it should show a message saying that.

borpin, 2019-08-21:
> The spinner loading for ever is a bug

I would not regard it as a bug, it is doing what is intended, telling you it is looking for data (as opposed to just sitting doing nothing). However, it is not very useful to a new user who may not have configured any data sources for EmonCMS to find.

gitmeister, 2019-08-21:
Then the FEEDS pane has a problem because it doesn't have a permanent spinner and instead shows a message pointing to the Feeds API Help just like all previous versions of EmonCMS.

pb66, 2019-08-21:
I would regard this as a bug too because the spinner should indeed spin to load the page, but if there is nothing to load it should say so. The previous pre-v10 inputs page displays this

[image]

the ever spinning spinner just suggests it's stuck, it isn't a good indicator of a fault or status, it just fills the time whilst loading.

borpin, 2019-08-21:
> The previous pre-v10 inputs page displays this

I was not aware that was displayed (never had a V9.x without data) but that is what should happen.

TrystanLea, 2019-08-21:
I've just fixed this in the master branch and will merge to stable soon

closed 2019-08-24 by TrystanLea