emoncms/emoncms #1405: min_value_allowed and max_value_allowed processes are reversed

pb66, 2019-09-11:
><p>If value is greater than <i>max value allowed</i> then the value passed to following process will be the <i>max value allowed</i></p>

If the value is greater than the "max value allowed" it should be blocked not "allowed" to pass!

likewise for the min value allowed

> <p>If value is lower than <i>min value allowed</i> then the value passed to following process will be the <i>min value allowed</i></p>

Both description are actually technically correct for the way they work, they just don't make sense!

https://github.com/emoncms/emoncms/blob/10.1.5/Modules/process/process_processlist.php#L849-L859

first committed in May 2018 https://github.com/emoncms/emoncms/commit/1ac0a218af81405c123df8471cf55ac134e222e5 

so we cannot change the existing functions as it will effect the behaviour of existing processlists.

To correct this we would need to create 2 new functions with new names (eg "min_allowed" and "max_allowed") and perhaps add the incorrect processes to the "hidden group" so they do not appear in the process selection dropdown?

    public function max_allowed($arg, $time, $value)
    {
        if ($value<$arg) $value = $arg;
        return $value;
    }
    
    public function min_allowed($arg, $time, $value)
    {
        if ($value>$arg) $value = $arg;
        return $value;
    }

pb66, 2019-09-11:
Actually whilst we are at it!

We should be using ">=" and "<=" since (for example) **minimum value allowed "10"** should pass 10 as well as anything greater than 10! Currently it passes anything less than 10, ie 9 and below, so even the **"value allowed"** part is incorrect in addition to the direction error.

```
public function max_allowed($arg, $time, $value)
{
    if ($value<=$arg) $value = $arg;
    return $value;
}

public function min_allowed($arg, $time, $value)
{
    if ($value>=$arg) $value = $arg;
    return $value;
}
```

chaveiro, 2019-09-11:
> Both description are actually technically correct for the way they work, they just don't make sense
Actually i see them as capping processes to limit vales positively or negatively.
Why do you think it does not make sense, is the naming of the processes ?

pb66, 2019-09-11:
> Actually i see them as capping processes to limit vales positively or negatively.

Ah yes, my bad!

I was thinking it was a filter, like the "allow positive" or "allow negative" processes but with a definable level ie allow maximum 10 means allow everything **10 and under**. 

These functions should perhaps be called "capped max" and "capped min" as they do not simply allow or not allow, they "cap". 

I'll have to have another look at this when I get a chance.

chaveiro, 2019-09-11:
Yes maybe the naming was not the best choice.

closed 2019-11-25 by TrystanLea