emoncms/emoncms #1422: mysql instructions for ubuntu install outdated

evilantishad0w, 2019-09-22:
Re: https://github.com/emoncms/emoncms/blob/master/docs/LinuxInstall.md

The instructions under "Create a MYSQL database" are outdated; instead of the command:

> mysql -u root -p

The following should be used:

>sudo mysql --user=root mysql

(Solution found at https://askubuntu.com/questions/763336/cannot-enter-phpmyadmin-as-root-mysql-5-7 )

borpin, 2019-09-22:
Quite possibly so - these instructions are badly in need of update although the intention is to provide a script and probably retire the various instructions.  The Install process that will be used by the script is here 

https://github.com/openenergymonitor/EmonScripts/blob/master/install/mysql.sh

borpin, 2019-10-01:
@evilantishad0w The new install script instructions look [like this](https://github.com/openenergymonitor/EmonScripts/blob/master/install/mysql.sh)

closed 2019-11-25 by TrystanLea