emoncms/emoncms #1424: Error on Inputs page

borpin, 2019-09-22:
This error just appeared in the browser window - the tab was in the background at the time.

[image]

It might be this line of code `devices[nodeid].inputs.push(inputs[z]);`

In console

[image]

alexandrecuer, 2019-09-23:
I also noticed that the input/view route does not work with microsoft Edge (but who use Edge nowadays?)
[image]

emrysr, 2019-09-25:
@borpin - I can't reproduce the error you reported... any additional information might allow me to reproduce it.

I've created a pull request #1428 that simply checks the variable type before running the function, this is always good practice - however this exception shouldn't occur as the variable is always set to be the expected type (array)

emrysr, 2019-09-25:
@alexandrecuer - thanks for reporting this issue, however I dont think this error is related to the original issue. I'll create a new github issue to track this problem

borpin, 2019-09-25:
I am getting random errors like this ATM. On a different error, @TrystanLea suggested one of the reasons was REDIS getting out of sync. Could that be a factor here?

borpin, 2019-10-11:
Errors seem to have gone so closing for now. Cheers.

closed 2019-10-11 by borpin