emoncms/emoncms #1436: Suggested modifications for default ini settings

borpin, 2019-10-01:
https://github.com/emoncms/emoncms/blob/6274f7979f89248a41847f5afd48b0a0a11491bb/default-settings.ini#L6

Add a comment that the items in [] denote sections. Settings must sit within the right section.

Perhaps the example settings should have all sections listed?

https://github.com/emoncms/emoncms/blob/6274f7979f89248a41847f5afd48b0a0a11491bb/default-settings.ini#L28-L29

Should this be in example as it is expected to be set to false - something for admin page?

https://github.com/emoncms/emoncms/blob/6274f7979f89248a41847f5afd48b0a0a11491bb/default-settings.ini#L43
Add a note that if the MQTT settings are changed, the service must be restarted.

https://github.com/emoncms/emoncms/blob/6274f7979f89248a41847f5afd48b0a0a11491bb/default-settings.ini#L74

Is 600 too long?

https://github.com/emoncms/emoncms/blob/6274f7979f89248a41847f5afd48b0a0a11491bb/default-settings.ini#L83-L85
These say experimental but are enabled by default.

https://github.com/emoncms/emoncms/blob/6274f7979f89248a41847f5afd48b0a0a11491bb/default-settings.ini#L173

Should the default smtp settings be blank?
I suggest these examples are put in the example ini but commented out.

https://github.com/emoncms/emoncms/blob/6274f7979f89248a41847f5afd48b0a0a11491bb/default-settings.ini#L196

It is quite likely the log level might be changed so I suggest putting in example.

TrystanLea, 2019-10-01:
thanks @borpin happy for these changes, do you think you could create a pull request with your suggestions?

borpin, 2019-10-02:
PR done #1437 

Not sure what to do about this one -

These settings say experimental but are enabled by default.
https://github.com/emoncms/emoncms/blob/6274f7979f89248a41847f5afd48b0a0a11491bb/default-settings.ini#L83-L85

TrystanLea, 2019-10-02:
not sure about mysqltimeseries and virtualfeed, @chaveiro are these default settings ok?

chaveiro, 2019-10-02:
Yes they are, you may remove 'experimental' from the comments, false is the recommended setting and gives better precision. 
Only set to true if you get performance issues with virtual feeds.

borpin, 2019-10-11:
@chaveiro I was going to update the comment, but I am not sure the comment actually makes sense.

```
; experimental feature for virtual feeds average, default is true, set to false to activate average agregation with all data points, will be slower
```

"set to false to enable"?

Should it be

```
; Feed Average feature. Default is false (disabled). 
; Set to True to activate average aggregation with all data points, will be slower.
```

chaveiro, 2019-10-11:
The text can be something like:
"Default to false. Data-sampling enable faster response at precision cost. Only set to true if you have performance issues. "
Other thing is that you shoudl only have a global **feed[data_sampling]** config, not  a config for each feed like: 
virtualfeed[data_sampling] = false 
mysqltimeseries[data_sampling] = false

borpin, 2019-10-12:
AIUI, False = Enabled and True = Disabled.

Is that correct?

closed 2019-11-25 by TrystanLea

borpin, 2019-11-26:
This has not been fixed. Still refers to an 'experimental setting'