emoncms/emoncms #1450: Bug: MQTT Topic in process list returned by API is escaped

borpin, 2019-10-11:
Inside the process list value strings returned by the API, the MQTT topic separator is escaped
```
"processList":"2:0.1,1:14,35:emoncms\/TH1IntTemp"
```
Should be
```
"processList":"2:0.1,1:14,35:emoncms/TH1IntTemp"
```

Would this process list import via the API?
```
emoncms/input/process/set?inputid=0&processlist=0:0
```

TrystanLea, 2019-12-12:
Thanks @borpin

TrystanLea, 2020-01-21:
This looks like a duplicate of the same escaped json issue here https://github.com/emoncms/emoncms/issues/977. 

I've marked that issue as an enhancement for now as the current behaviour doesn’t strictly break anything in emoncms, it is still parsed correctly when used, it's just that it would be nicer if it did not escape in this way.

closed 2020-01-21 by TrystanLea