emoncms/emoncms #1452: Processing of Envvars is disabled

borpin, 2019-10-15:
This line is commented out

https://github.com/emoncms/emoncms/blob/7f12b5c350ef3999d33d7da0d203e9dd54905978/process_settings.php#L35

I think you should be able to check for an Envvar within the merge

https://github.com/emoncms/emoncms/blob/7f12b5c350ef3999d33d7da0d203e9dd54905978/process_settings.php#L79-L79

If you simply have a function to check for the `{{` at this point and either return the original if not found, or return the corresponding envvar.

No need to iterate over the settings a second time - let the recursion do its job.

borpin, 2019-10-15:
Real quick start - can't work out how to get the error output to tell me where it is getting to!

```php
function check_envvars ($var){
    global $error_out;

    if (is_string($var)) {
        if (strpos($var, '{{') !== false && strpos($var, '}}') !== false) {
            preg_match('/{{(.*)}}/', $var, $match);
            if (!isset($_ENV[$match])) {
                $error_out .= "<p>Error: environment var '${match}' not defined in config section [${section}], setting '${key}'</p>";
            } else {
                $error_out .= "<p>Error: environment var ";

                // $newval = str_replace('{{'.$match.'}}', $_ENV[$match], $value);

                // // Convert booleans from strings
                // if ($newval === 'true') {
                //     $newval = true;
                // } else if ($newval === 'false') {
                //     $newval = false;

                // // Convert numbers from strings
                // } else if (is_numeric($newval)) {
                //     $newval = $newval + 0;
                // }

                // // Set the new value
                // $config[$section][$key] = $newval;
            }
        } else{
            return $var;
        }
    } else {
        return $var;
    }
}
```

Sorry, all I have time for right now @TrystanLea

takkaria, 2019-10-15:
Maybe this would work?  Untested

```php

function resolve_env_vars($input) {
    if (!is_string($input)) {
        return $input;
    }

    // Find {{ }} vars and replace what's within them with the
    // named environment var
    if (strpos($input, '{{') !== false && strpos($input, '}}') !== false) {
        preg_match_all('/{{([^}]*)}}/', $input, $matches);
        foreach ($matches[1] as $match) {
            if (!isset($_ENV[$match])) {
                echo "<p>Error: environment var '${match}' not defined</p>";
            } else {
                $processed = str_replace('{{'.$match.'}}', $_ENV[$match], $input);

                // Convert booleans from strings
                if ($processed === 'true') {
                    $processed = true;
                } else if ($processed === 'false') {
                    $processed = false;
                // Convert numbers from strings
                } else if (is_numeric($processed)) {
                    $processed = $processed + 0;
                }

                // Set the new input
                return $processed;
            }
        }
    }

    return $input;
}
```

borpin, 2019-10-16:
`$match` is an array.

This shouldn't need the `str_replaced` bit now I don't think, just return the variable if set.

However, I could not work out how to set an ENV variable for PHP to read (I tried to set the log_level)!!

```php
    preg_match('/{{([^}]+)}}/', $input, $match);

    if ($value = getenv($match[1])) {
        return $value;
    } else {
        error_log ("************ environment var '$match[1]' NOT found");
        return $input;
    }
```

@Trystan, I note that on reloading a page, process settings is being called multiple times!

borpin, 2019-10-16:
Showing my lack of PHP...

This won't work if the value of the environment variable is false 😢 

```
if ($value = getenv($match[1]))
```

borpin, 2019-10-16:
I've had a bit of fun this morning and learnt some PHP to boot 😀 (coments on style etc gratefully received!

First, I think the `paarse_ini_file` should use the `INI_SCANNER_TYPED` directive
```
    $CONFIG_INI = parse_ini_file("default-settings.ini", true, INI_SCANNER_TYPED);
    $CUSTOM_INI = parse_ini_file("settings.ini", true, INI_SCANNER_TYPED);
```

Second call the function in such a way that if the environment var is not found, it falls back to the default (though I'm not 100% convinced this is right but it seems to work!).

```
$defaults[$k] = resolve_env_vars($v,$defaults[$k]);
```

Used getenv as it is case insensitive. Re my bit above, `getenv` always returns a string it seems.

```
function resolve_env_vars($input, $defaultval) {
    // To do any processing we need a string, and a matching pair of {{ }}
    // Otherwise just return what we got.
    if (!is_string($input) ||
            strpos($input, '{{') === false ||
            strpos($input, '}}') === false) {
        return $input;
    }

    preg_match('/{{([^}]+)}}/', $input, $match);

    if ($value = getenv($match[1])) {
        // Convert booleans from strings
        if (strcasecmp($value, "true") == 0) {
            $valueout = true;
        } else if (strcasecmp($value, "false") == 0) {
            $valueout = false;
        // Convert numbers from strings
        } else if (is_numeric($value)) {
            $valueout = $value + 0;
        } else {
            $valueout = $value;
        }
        return $valueout;
    } else {
        error_log ("************ environment var '$match[1]' NOT found for '$input' - setting to default '$defaultval'");
        return $defaultval;
    }
}
```

borpin, 2019-10-16:
A single page reload calls `process_settings.php` 4 times I think.

For instance a debug line in the `merge_ini`

```
[Wed Oct 16 07:59:54.215892 2019] [php7:notice] [pid 9769] [client 192.168.7.123:59760] **** update default '_DB_USER_'
[Wed Oct 16 07:59:54.946091 2019] [php7:notice] [pid 9813] [client 192.168.7.123:59762] **** update default '_DB_USER_', referer: http://192.168.7.77/admin/view
[Wed Oct 16 07:59:54.947399 2019] [php7:notice] [pid 9773] [client 192.168.7.123:59761] **** update default '_DB_USER_', referer: http://192.168.7.77/admin/view
[Wed Oct 16 07:59:54.948475 2019] [php7:notice] [pid 9769] [client 192.168.7.123:59760] **** update default '_DB_USER_', referer: http://192.168.7.77/admin/view
```

takkaria, 2019-10-16:
I don't think it makes any sense to fall back onto a default if an env var can't be found.  It makes a lot more sense to make it a fatal error I think.

I'm going to suggest the following function. It will handle things like `{{ENV_VAR}}_test_{{ENV_VAR2}}` which can be useful.  @borpin your most recent version ignores this kind of templating - so even if someone wrote `test_{{ENV}}`, if ENV was `fred` you'd just get back `fred`, not `test_fred`.  I've also added back in the loop to handle multiple env var replacements.

```
function resolve_env_vars($value) {
    // To do any processing we need a string, and at least one pair of {{ }}
    // Otherwise just return what we got.
    if (!is_string($value) ||
            strpos($value, '{{') === false ||
            strpos($value, '}}') === false) {
        return $value;
    }

    preg_match_all('/{{([^}]+)}}/', $value, $matches);
    foreach ($matches[1] as $match) {
        $env_name = $match;
        $env_value = getenv($env_name);
        if ($env_value === false) {
            echo "<p>Error: environment var '${env_name}' not defined</p>";
            return $value;
        }

        $value = str_replace('{{'.$env_name.'}}', $env_value, $value);
    }

    // Convert booleans from strings
    if ($value === 'true') {
        $value = true;
    } else if ($value === 'false') {
        $value = false;

    // Convert numbers from strings
    } else if (is_numeric($value)) {
        $value = $value + 0;
    }

    // Set the new value
    return $value;
}
```

borpin, 2019-10-16:
@takkaria, no bother.

One point, I found `($value === 'true')` to not work if the case used was not 'exact'. As it is a string being returned I felt the string compare was more robust. YMMV

> @borpin your most recent version ignores this kind of templating

Yes, not something I was familiar with.

Can you offer some test cases to test this templating?

@TrystanLea How would this work if the setting value itself is an Array in it's own right. I'm thinking the `engines_hidden` setting. May be a flaw in the main `merge_ini`.

takkaria, 2019-10-16:
I guess I just expect it to work "like string templating I've used in other places", which isn't very useful...

* `mysql://{{USER}}@{{PASS}}:{{HOST}}/` should work
* If I set `BOOLEAN_SETTING=true` then `{{BOOLEAN_SETTING}}` should be (boolean) `true`
* If I set `TIMEOUT_SECS=600` then `{{TIMEOUT_SECS}}` should be (int) `600`

That's all I can think of.  (Case noted with strcasecamp, I amended the version above.)

relnah, 2019-10-19:
`} else if (strcasecmp($value, 'false' == 0) {`
is missing a ) after 'false'.

borpin, 2019-10-25:
@takkaria @relnah are you in a position to test this?

relnah, 2019-10-25:
Yes, I can clone your branch and base my docker build on it. Not sure I have real data to test all cases mentioned above. But some at least.

takkaria, 2019-10-25:
Nope, I'm not anymore, sorry!

relnah, 2019-11-17:
@TrystanLea Sorry for delay. As @borpin said, I tested and commented in issue where the actual PR is :)
Is still working well for me.

TrystanLea, 2019-12-12:
This is now merged, thanks all for your help getting this to work!

closed 2019-12-12 by TrystanLea

emrysr, 2019-12-14:
while working on https://github.com/emoncms/emoncms-docker I found it useful to see what options were being set via env variables. This is a module I use to show the 'origin' of a setting:
https://github.com/emrysr/emoncms-debug

[image]

while making this module I noticed that the function `ini_check_envvars()` had been removed from `process_settings.php`.  This pull request (https://github.com/emoncms/emoncms/pull/1483) puts it back and allows for the environment variables to be modified via the php global `$_ENV`

borpin, 2019-12-14:
I don't think you had updated your branch as that function is not used anymore.