emoncms/emoncms #1453: openEVSE emoncms app Type Error

gr0mit1, 2019-10-16:
Hi
I've got this error when selecting the graph and zooming to a particular time period.  Any ideas on what is causing it?
[image]

TrystanLea, 2019-12-12:
Thanks @gr0mit1 noted as a bug, we will look into it

TrystanLea, 2020-01-20:
Hello @gr0mit1 we are removing this app for now as its a replica of My Electric 2, the intention was to add specific OpenEVSE features but until that point we've removed it to avoid confusion and focus on the My Electric 2 app. https://github.com/emoncms/app/commit/5d43ba90f6c2bfbae7eeace576305b7e77148eef

closed 2020-01-20 by TrystanLea