emoncms/emoncms #1456: low write mode causing problems

Hecot, 2019-10-21:
I've followed the guide step by step to activated the low write mode in emoncms but now the webpage is not available anymore and I'm getting a http error 500

[image]

I've double checked the steps and found that the feedwriter service is not running ...

[image]

I don't know if this is causing the issue and how to solve this? can someone please help ... I'm not the RPi pro ... sorry!

Hecot

borpin, 2019-10-21:
Hi @Hecot, best place to ask is on the community https://community.openenergymonitor.org/

Cheers

Hecot, 2019-10-21:
@borpin Thanks - did a post there too!

borpin, 2019-10-21:
> did a post there too!

Ok, I'll have a look. Can you close this issue please?

borpin, 2019-10-21:
Actually don't close - @TrystanLea another guide for the archive please.

Hecot, 2019-10-21:
Please update instead of archive!!!

closed 2019-11-25 by TrystanLea