emoncms/emoncms #1462: $path_to_config = 'settings.php';

borpin, 2019-10-24:
https://github.com/emoncms/emoncms/blob/1f3e991e48cc60416d19d698b4e80f0a9953de26/Modules/admin/admin_controller.php#L54

@TrystanLea Is this a problem now we have moved to an ini file?

TrystanLea, 2019-12-12:
Thanks @borpin noted for next development itteration

emrysr, 2019-12-13:
probably just a check that the new `.ini` files exist and referencing the `.php` version if not.
```
    $path_to_config = file_exists('settings.ini') ? 'settings.ini' : 'settings.php';
```
not sure what should be shown if both files exists? the`.php` file takes precedents in the `process_settings.php` file:
https://github.com/emoncms/emoncms/blob/1a587503d00780214640f70c7cc4e17bdf43ab27/process_settings.php#L20
https://github.com/emoncms/emoncms/blob/1a587503d00780214640f70c7cc4e17bdf43ab27/process_settings.php#L31

borpin, 2019-12-14:
I don't think that is the right logic. I *think* the logic @TrystanLea employed was that the `settings.php` trumps the ini settings system. Otherwise a pull could revert to default settings.

Also on a new card, if a backup is done from an old system, it will import a settings.php.

I could be wrong - we did discuss it.

TrystanLea, 2019-12-15:
$path_to_config was used for editing the log level in the settings file. This feature is currently disabled. I think when we do re-enable it, it should only be available for the settings.ini file, so if we detect that the settings.ini file is not present we just disable the whole feature.

borpin, 2019-12-15:
That sounds good to me.

TrystanLea, 2020-01-20:
I've fully disabled the code associated with this settings for now and am changing this issue to be an enhancement rather than a bug. https://github.com/emoncms/emoncms/commit/a748a42d985c7b8667f0c95cfc081c78caca4964

The enhancement is to add a settings.ini editor

TrystanLea, 2023-04-06:
closing as similar issue open elsewhere

closed 2023-04-06 by TrystanLea