emoncms/emoncms #1477: Editrealtime delete data in window fails

borpin, 2019-11-26:
Trying to use the 'Delete Data In Window' fails with following error

[image]

Related to #1011 & #1006

alexandrecuer, 2019-11-26:
Hi
I had tested a lot....in most cases, it works but there are cases where it does not work when it does not find properly the starting point...
Anyway did not find a more proper way to code the thing....i will check again....
The message seems to be a frontoffice one rather than coming from the API
Do yu have the same error with all timeranges ?
Best
Alex

borpin, 2019-11-26:
Yes it seems all time ranges. I cannot get it to select a different multiple - once it has decided to use a time ending in '1' say, it continues to use that multiplier.

Could the delete be done by manually specifying the time range?

I think actually displaying the data in the window may well help (same as data viewer).

Cheers

borpin, 2019-11-26:
The other problem is missing data.  Looking at a feed now that I know has missing data, and that is not shown (as missing) in the Vis window.

It would lead to attempting to delete data that did not exist.  Equally if you knew it was 'null' you could potentially add the data item in and fill in data.

Another though, it would be useful to have the same hover tip on the Vis that appears on the Data Viewer (time, timestamp data etc).

Cheers

alexandrecuer, 2019-11-26:
sorry sorry sorry  I answered too fast....the "Delete Data In Window" button never worked...

Actually what I implemented is just under this button (see the orange scribble on the screenshot) and it is working in version 10.1.6 as far as I am concerned
The backoffice function concerned is in the PHPFina engine, and is called scalerange 
I created the code in january 2019
cf https://github.com/emoncms/emoncms/commit/92d94660ed1aa1d739c2f8a8cf9096d07ea0e909#diff-5b2bae4bf00f1938393368515a475c88

To use it, just type NAN if you want to erase all the window
[image]

after the clic, you will see the window erased...everything or nearly everything in the timerange replaced by NAN in the HEXA feed...
[image]
usefull if you have bugs or erroneous values
initially I wanted a basic scaling function but it was easy to code an erasing functionnality and as the "Delete Data In Window" did not work, I did both things in one shot 

I agree it is very well hidden :-)

if yu dont find the zone, use the fullscreen mode

tell me if OK or not

best

Alex

borpin, 2019-11-27:
If the feature does not work, then I think it should be removed.

Is there no way to get the data points to sit on the right boundary?

alexandrecuer, 2019-11-28:
what actually the function does to find the boundaries is :

`` $start = intval($start/1000);``
`` $end = intval($end/1000);``

`` $start=floor($start/$meta->interval)*$meta->interval;``
`` $end=floor($end/$meta->interval)*$meta->interval;``

``$pos_start=4*floor(($start-$meta->start_time)/$meta->interval);``
`` $pos_end=4*floor(($end-$meta->start_time)/$meta->interval);``
`` $nbwrites=($pos_end-$pos_start)/4;``

$start and $end are graphically choosen by the user, they are extracted from the timerange of the window and are unix time stamps
They do not necessarily correspond to real points in the feed, so we have to find the closest ones, and that's why we have to calculate the addresses (or positions) $pos_start and $pos_end

So personnaly I don't see a more proper way to get the data points exactly sitting on the real boundary, if I understand correctly yur question....All this is a consequence of sampling and we need to approximate...
if you have a better solution, it could be nice :-)

borpin, 2019-11-28:
> $start and $end are graphically chosen by the user

But it is impossible to 'graphically' choose a precise start point or to know what an actual data point timestamp is to set a correct boundary time.

For the user, the key use case is to edit data spikes so they want the precise data point to edit.

I think the code should take that graphical start and then look for the closest actual data point to the graphical start (is there a modulo function?) and work from there adjusting the zoom so that all the sampled data points are real data (so the window is then a multiple of data interval).

Obviously it might be impossible to have **all** the points for the window selected (hence sampling), but that should be clear to the user - a message stating the data has been sampled.

The user can then zoom in to a point such that all the datapoints are visible.

It would be useful to have the actual datapoints listed as CSV in a window below like the Graph module does.

borpin, 2019-11-28:
> $start=floor($start/$meta->interval)*$meta->interval;

I **think** you could use (totally untested...)
```
 $start=$start - ($start % $meta->interval);
```
to give a start on an interval boundary.

borpin, 2019-11-28:
How about this
```php
// convert window start to seconds
$start = intval($start/1000);
//how long is window start since meta start (does anegative need to be accounted for?)
$time_from_meta_start = $start – $meta->start;
// time from meta start to start as an exact multiple of meta intervals (delete the modulus)
$interval_time_from_meta_start = $time_from_meta_start – ($time_from_meta_start % $meta->interval);
$pos_start = $meta->start + $interval_time_from_meta_start;
```

borpin, 2019-11-29:
I've now dug around the code and I can see what it is doing but not always how - my PHP and JS is not that good!

I think *a* solution is to set the `timeWindow` as a multiple of the feed interval and set the start on a data boundary.

https://github.com/emoncms/emoncms/blob/140e7e2cdbc1172d3f480c6fb42423a6eab8eb9b/Modules/vis/visualisations/editrealtime.php#L87-L87

I cannot see how to get the interval. I'm sure it must be able to be extracted by the controller, I just cannot see how!

If I could get the interval, I'm sure we should be able to set the start correctly (as in on a boundary).

alexandrecuer, 2019-11-30:
OK nice it can be an idea :-)
But honestly, dont yu think the current implementation is enough for 95% of the cases ?
Anyway the js code in the view is the following (lines 151 of the editrealtime.php file) :
```
  $('#multiply-submit').click(function () {
    var multiplyvalue = $("#multiplyvalue").val();
    $.ajax({
      url: path+'feed/scalerange.json',
      data: "&apikey="+apikey+"&id="+feedid+"&start="+start+"&end="+end+"&value="+multiplyvalue,
      dataType: 'json',
      async: false,
      success: function() {}
    });
    vis_feed_data();
  });
```
so you give to the php API the following params : apikey, id, start, end and multiplyvalue
if multiplyvalue is NAN, then the API will erase from start to end
timewindow is only used when the user arrives on the page, in order to have a default timerange 

as we are in js, start and end are expressed in  milliseconds since 1 January 1970 00:00:00
so the interval in milliseconds is end-start

but I may not have understood your request correctly.

borpin, 2019-11-30:
No I am going back to the definition of the timewindow and I suggest setting the start of that window to be set relative to the feed start time plus a multiple of the feed interval.

To do that we need the **feed start time and the interval** at this point in the code;

https://github.com/emoncms/emoncms/blob/140e7e2cdbc1172d3f480c6fb42423a6eab8eb9b/Modules/vis/visualisations/editrealtime.php#L82

so you can define 
```
var feed_interval = xyz
var feed_starttime = 123
```

Then `timewindow` is calculated to be
```
feed_starttime + (feed_interval*<num of intervals>)
```
where the `num of intervals` takes you to the nearest feed boundary to the selected timewindow start.

If you then pass the functions to get the data a start point that is a boundary and an interval value that is a multiple of the feed interval, all the data returned **should** (I think) be valid data points for editing.

This is the values I need to try it;
* Feed start time
* feed interval

alexandrecuer, 2019-11-30:
OK if you want to work on those datas in js, you will need to follow the following route, assuming you work on feed 1 :
feed/getmeta.json?id=1

it will return :
- interval in s (for ex 300 if set to 5 minutes)
- start_time in s (for ex 1520527800)
- npoints
all this coming from mariadb

anyway, this is not usually realised in the frontoffice but within the engine functions...
in the engine, you've got a meta field which gives you all that :
$meta->interval 
$meta->start_time
cf https://github.com/emoncms/emoncms/commit/92d94660ed1aa1d739c2f8a8cf9096d07ea0e909#diff-5b2bae4bf00f1938393368515a475c88

borpin, 2019-11-30:
> anyway, this is not usually realised in the frontoffice but within the engine functions...

Yes I realise that. I just cannot work out how to expose it in the same way as, for instance, `$feedid` is set os it can be used in the Javascript function.

alexandrecuer, 2019-11-30:
The 
``
var feedid = "<?php echo $feedid; ?>;
``
You should better user an ajax request in asynchrone mode....
Is not a very good practice

borpin, 2019-11-30:
yes just started looking at that and realised the `get_feed_meta` api call is not correct - it needs a feedID.

borpin, 2019-11-30:
So, I seem to have got it working here but on limited test data!

in `editrealtime.php`
```
  var timeWindow = (3600000*24.0);        //Initial time window BPO shortened the default to a day
  var start = ((new Date()).getTime())-timeWindow;    //Get start time
  var end = (new Date()).getTime();       //Get end time
  var feed_meta = get_feedid_meta(feedid,apikey);  //get feed meta data BPO check for feed type?

  vis_feed_data();

  function vis_feed_data()
  {
    start_time = feed_meta.start_time * 1000; //in miliseconds
    feed_interval = feed_meta.interval * 1000; // in miliseconds

    var time_from_meta_start = start - start_time;
    var interval_time_from_meta_start = time_from_meta_start - (time_from_meta_start % feed_interval) + feed_interval;
    // modify start time so it sits on a feed data boundary
    start = start_time + interval_time_from_meta_start;
    // modify end so it sits on a feed data boundary
    end = end - (end % feed_interval);
    
    var npoints = 800;
    interval = Math.round(((end - start)/npoints)/1000);
    // make the interval a multiple of the feed interval
    interval = interval - (interval % feed_meta.interval);
    // do not ignore missing data, limit to interval
    var graph_data = get_feed_data(feedid,start,end,interval,0,1);
    var stats = power_stats(graph_data);
```
and in `api.js` (though I think the `get_feed_meat` should be corrected)
```
// Get meta
function get_feedid_meta(feedID,apikey){
  var meta = [];
  var query = "id="+feedID;
  if (apikey!="") query+= "&apikey="+apikey;
  $.ajax({                    
    type: "GET",
    url: path+'feed/getmeta.json',
    data: query,  
    dataType: 'json',
    async: false,
    success: function(dt) { meta = dt; }
  });
//  meta = JSON.parse(meta);
  return meta;
}
```

I have noticed that zooming in and out can break it (so not data is shown), but I'm not sure if that was an existing bug.

I have also included points where there is no data - important if you are trying to edit the data.  This could be added as an option.

Thoughts?

borpin, 2019-11-30:
> You should better user an ajax request in asynchrone mode....

I note all the API calls are **not** in async mode.

alexandrecuer, 2019-11-30:
Yes for that one
But all :-)..
check some modules for example sync or postprocess
Yu will see ajax calls in asynchrone mode
Anyway it is ok here because request are fast...

borpin, 2019-11-30:
> But all :-)..

I only did a quick check :)

I also noticed some of the functions check for, or use, an apikey, but no key is passed. Is that a global variable then? If not, is this an error?

alexandrecuer, 2019-11-30:
no it's not a bug, once you are logged in you dont need the apikey....it is stored in redis
actually, the key var was intended for people embedding graphics in external websites...dont know if a lot of people have such a practice :-)

there is a lack of readme/documentation in the vis module I think but OEM guys are rather concentrating on the graph module....I had started one while trying to improve the lisibility of multigraph.js
see that PR :
https://github.com/emoncms/emoncms/pull/1427/files

TrystanLea, 2020-01-20:
I've implemented pretty what you suggested @borpin and can confirm its now working much better for phpfina, I've also made a small change to improve support for phptimeseries (limitinterval set to 0 results in actual phptimeseries timestamps returned) 

https://github.com/emoncms/emoncms/commit/67419fccbc35eaab401f706da62cb35a656acd8e

closed 2020-01-20 by TrystanLea