emoncms/emoncms #1478: Consider URL parameter to load dashboards with sidebar collapsed

TrystanLea, 2019-11-27:
https://community.openenergymonitor.org/t/option-to-minimize-the-setup-sidebar/12583

TrystanLea, 2019-12-12:
Noting as feature request to be kept open as a reference for future development, when time/resources allow.

emrysr, 2019-12-13:
might be an idea to always show the close icon shown on mobile devices:
[image]

and saving the selection as a cookie value that is remembered between page loads??  might be useful for desktop users... usability might need some testing on mobile devices/

borpin, 2019-12-23:
@emrysr The user who raised this uses smart TV so no mouse function available.

His suggestion is for it to be an option for a dashboard, which seems like a good idea as that is where it is most likely to be used.

Perhaps an option to hide the top bar as well.

emrysr, 2019-12-23:
here's a pull request with a dashboard "fullscreen" option...
https://github.com/emoncms/dashboard/pull/245

closed 2020-01-22 by TrystanLea