emoncms/emoncms #1479: Mqtt_input service down: ‘activating (auto-restart)’ state

fcattaneo2, 2019-12-08:
new installation, Raspberry BUSTER + emoncms  10.1.13.

In admin dashboard, mqtt_input service is in ‘activating (auto-restart)’ state
Loaded: loaded (/etc/systemd/system/mqtt_input.service; enabled)
       Active: activating (auto-restart) (Result: exit-code) since Sat 2019-11-23 16:59:36 CET; 7s ago
      Process: 27734 ExecStart=/usr/bin/php /var/www/emoncms/scripts/phpmqtt_input.php (code=exited, status=255)
     Main PID: 27734 (code=exited, status=255)

For fix the problem i change the line 46 in /emoncms/scripts/phpmqtt_input.php from :

chdir(dirname(__FILE__)."/../../../");

to
    chdir(dirname(__FILE__)."/../");

Ciao.

borpin, 2019-12-08:
As I said on the community forum, I suspect this is because of the way you installed the system. It works for everyone else.

https://community.openenergymonitor.org/t/mqtt-input-service-down-activating-auto-restart/12666

borpin, 2019-12-08:
I'll respond on the community, but you have installed the wrong service.  This one is deprecated.

closed 2019-12-12 by TrystanLea