emoncms/emoncms #1485: Device Already Exists Error

borpin, 2019-12-16:
https://github.com/emoncms/emoncms/blob/ccb3a41ed17879ff154dfcabf7fa58d010ef8dce/Modules/input/Views/input_view.js#L878

Getting these errors on a regular basis.  Could be for any existing node.

[image]

It has also been reported on the community.

https://community.openenergymonitor.org/t/constant-device-already-exists-error-alert-poup/12714

Whilst clearing REDIS cache does fix it, it does not solve why it is happening.

The code has reached this point, because it does not think the device exists, yet, when it checks again, it does exist.

Something not right......

TrystanLea, 2020-01-21:
Even though I've seen this issue myself, I just cant replicate it, I will have to keep it open until I see it again and debug then.

TrystanLea, 2020-02-06:
Noticing this issue again after upgrading and SD card. I realise that the issue is created by a node posting to emoncms via emoncms_mqtt creating a redis entry for the device. The device module then fails to load the full list of devices as there is already a single entry in the redis cache.

To fix this issue the redis cache is reloaded before adding the new device to the redis cache
https://github.com/emoncms/device/commit/d5b0a565d339547ca8a2bc346c4cc0f50aef9870

Restoring the SD card again did not show the error but the issue is intermittent so more testing is required to confirm that this does fully fix this issue

borpin, 2020-02-06:
If it appears to be stable, can you either create a branch for testing or else merge it to master so there can be wider testing?

I think it might be an idea to post this to the forum as, TBH, it is one of those I just ignore and I suspect so do many others!

It would be great if this bug has been squashed!

TrystanLea, 2020-03-27:
Already merged to master, we can reopen if still an issue

closed 2020-03-27 by TrystanLea