emoncms/emoncms #1487: Power to kWh feed not resetting upon 'clear'.

danbates2, 2019-12-18:
At the delete feed dialog box, including clear and trim options, the clear option doesn't reset the power to kWh accumulator. Possibly this behaviour would apply to other accumulators.
It'll be a matter of preference as to whether the accumulator is reset, but I think if the aim is to 'clear' the feed, the the accumulated value should also be reset.
Trim may not reset the accumulator if deleting history is what one wants.

emrysr, 2019-12-18:
The previous value must be cached for the accumulation to happen??... Thanks @danbates2 I'll look into this.

danbates2, 2019-12-19:
Not sure. Just seems like common sense that if the desired action is to 'clear' the feed (the same as resetting!) then the value should be reset.
If clearing the feed is desired without the value reset, then the 'trim' can be used. Makes sense?

TrystanLea, 2020-01-20:
This is now fixed https://github.com/emoncms/emoncms/commit/230bd485bc77f073ed3489119b8381b69dd9c6ae

closed 2020-01-20 by TrystanLea