emoncms/emoncms #1494: Include file_exists check in try catch block

borpin, 2020-01-06:
Re discussion https://community.openenergymonitor.org/t/need-help-with-first-emoncms-setup-on-synology-diskstation/12439/22

The check for file exists may raise an exception.  The check should be included in a try catch block.

https://github.com/emoncms/emoncms/blob/1a587503d00780214640f70c7cc4e17bdf43ab27/Modules/admin/admin_model.php#L318

This should be included across all system info checks as we cannot assume any particular hardware/software setup.

tehnoinstyle, 2020-01-06:
those. i need to add this:
```
/usr/sbin/mosquitto
```
in:
```
/tmp:/var/services/tmp:/var/services/web:/var/services/homes:/var/lib/phpfiwa:/var/lib/phpfina:/var/lib/phptimeseries:/var/log/emoncms
```
like this:
```
/tmp:/var/services/tmp:/var/services/web:/var/services/homes:/var/lib/phpfiwa:/var/lib/phpfina:/var/lib/phptimeseries:/var/log/emoncms:/usr/sbin/mosquitto
```
but on my path:
```
/usr/sbin/mosquitto
```
not
```
mosquitto
```
[image]

borpin, 2020-01-06:
The safest route is to always do a try catch when checking for system files that EmonCMS has no control over.

tehnoinstyle, 2020-01-06:
those. i need to add this path?

tehnoinstyle, 2020-01-06:
add
```
/usr/sbin/mosquitto
```
did not solve the problem with error on line 318.

But new entries in the emoncms.log file did not appear.

emrysr, 2020-01-06:
hi @tehnoinstyle, have you tried the pull request above? is your previous comment relating to you trying that PR?

tehnoinstyle, 2020-01-06:
not, no trying that PR.

borpin, 2020-01-06:
> not, no trying that PR.

It should fix the issue.

tehnoinstyle, 2020-01-06:
from Modules/admin/admin_model.php delete:
```
public static function mqtt_version() {
        $v = '?';
        if(strtoupper(substr(PHP_OS, 0, 3)) === 'WIN') {
            $v = "n/a";
        } else {
            if (file_exists('/usr/sbin/mosquitto')) {
                $v = exec('/usr/sbin/mosquitto -h | grep -oP \'(?<=mosquitto\sversion\s)[0-9.]+(?=\s*)\'');
            }
        }
        return $v;
    }
```
and paste:
```
public static function mqtt_version() {
                }
            } catch (Error $e) {
                global $log;
                $log->info(_('Issue checking /usr/sbin/mosquitto').' - '.$e->getMessage());
                $v = "n/a";
            }
        }
        return $v;
```
yes or no?

emrysr, 2020-01-06:
hi @tehnoinstyle,
The above cut and paste examples are not complete... see the code in green here fore the full version: https://github.com/emoncms/emoncms/pull/1495/files

__Don't overwrite the master branch...__
If you create a new git branch `( checkout -b )` and pull in my changes in the new branch you won't need to overwrite the master branch and the above conflict will not appear.
```
git checkout -b emrysr-feature/issue-1494-file_exists-in-admin-model master
git pull https://github.com/emrysr/emoncms.git feature/issue-1494-file_exists-in-admin-model
```

this will allow you to revert back to the master branch if there are any issues:
` git reset master `

PR #1495 (2020-01-12): added try/catch block arround file_exists() code
outputted errors to log

fix #1494 

@tehnoinstyle / @borpin - I was unable to recreate the error that causes the Exception, however I have added the check based on the messages reported in the issue.

please let me know if this doesn't fix the errors you see.
closed 2020-01-12 by TrystanLea