emoncms/emoncms #1499: Logging, invalid login does not show IP address in log, which is needed for fail2ban.

xekon, 2020-01-15:
Logging, invalid login does not show IP address in log.

If I purposely try logging in with a bad password the log gives a result like this:
```
2020-01-15 12:09:53.552|INFO|index.php|user/login.json
2020-01-15 12:09:53.553|INFO|rememberme_model.php|login
2020-01-15 12:09:53.553|INFO|rememberme_model.php|getCookieValues: not present
```
Is there a way this could have the IP address added to the log? then I could create a fail2ban filter for this.

emrysr, 2020-01-20:
hi @xekon, could you please test this pull request with your setup and comment on the pull request if it returns the correct values in your log file. Seem that php might have different values for the client's IP address and might be incorrect in some cases..?

TrystanLea, 2023-04-06:
this was fixed here: https://github.com/emoncms/emoncms/blob/master/Modules/user/user_model.php#L464