emoncms/emoncms #1500: Virtual feed with same node name causes feed list error

TrystanLea, 2020-01-16:
See: https://community.openenergymonitor.org/t/dashboard-loggedin-vs-public/11626/19

I have not yet been able to replicate.

fillycheezstake, 2020-01-21:
I've added to the post in the link above.  Let me know if there's any further testing you'd like me to do.

borpin, 2020-01-24:
Note user feedback here

https://community.openenergymonitor.org/t/dashboard-loggedin-vs-public/11626/27

If user TZ is not found, then user browser TZ seems most logical.

emrysr, 2020-03-02:
https://github.com/emoncms/emoncms/pull/1522

TrystanLea, 2020-03-04:
fixed here: https://github.com/emoncms/emoncms/commit/ab04267ca1bcacdfc293db5e0647fa84a38ddb66

closed 2020-03-04 by TrystanLea

chaveiro, 2020-03-04:
See my comment here:
https://github.com/emoncms/emoncms/commit/ab04267ca1bcacdfc293db5e0647fa84a38ddb66#r37643142