emoncms/emoncms #1514: Functions do not do any error checks

borpin, 2020-01-25:
https://github.com/emoncms/emoncms/blob/d2e08a9e9a119681a0ec83d5850c18c8a15e3489/Modules/process/process_processlist.php#L844

https://community.openenergymonitor.org/t/upgrading-from-low-write-v9-8-30-to-latest-anything-to-watch-out-for/11797/21

A user had an exception because the function does not check for a valid set of arguments.  As a minimum, I think these functions should have a try catch. What a valid response would be though on error.... Ideally the Inputs page would show `error` where it might usually say `inactive`.

chaveiro, 2020-01-25:
And they should not have any.
Try catchs are expensive. 
There os at least one already on the execution pipeline, above the execution of functions.

borpin, 2020-01-25:
Long way above and the result means it is unclear to a user what the issue might be. 

You could do some basic checks for integrity at some point?

Perhaps the mqtt input should check at least for a numerical value.

TrystanLea, 2020-03-19:
Perhaps checking if $value is numeric would be sufficient in this case.
If not numeric it could just return the $value unmodified.

TrystanLea, 2020-03-19:
or properly filtering inputs so that they are all numeric, though if we where to support non numeric inputs that might be quite useful..

closed 2023-02-10 by borpin