emoncms/emoncms #1516: messy output after auth timeout

bwduncan, 2020-02-06:
Hey,

Every so often emoncms decides to log me out. That's annoying but the real frustration is that it doesn't tell me clearly. I'll be looking at a graph and I just get a request error:

> Request error: Received data not in correct format. Check the logs for more details

This could mean anything! In this specific case I think there is a bug in how feed_controller.php handles the auth error, because it returns this:

`<br />
<b>Notice</b>:  Undefined variable: results in <b>/var/www/emoncms/Modules/feed/feed_controller.php</b> on line <b>150</b><br />
{"success":false,"message":"Username or password empty"}`

so the Firefox console just has a JSON parse error.

I'm afraid I don't speak enough PHP to fix this, but I hope it's a simple fix.

Thanks,
Bruce

borpin, 2020-02-06:
This is a recurring issue identified. Yes it's a pain it is not clear you have been logged out. I think the page should auto refresh to the login page.

I'd like the 'remember me' to be permanent or even a second option to do that, or be able to set the user to never be logged out.

bwduncan, 2020-02-07:
I had a look through the code last night. Nothing jumps out at me as being obviously wrong which is frustrating. Can anyone fix the error above?

The "remember me" option should be flexible enough in principle to allow permanent logins, even across server reboots. The limit is set in the code at 90 days which is sensible from a security point of view, but tokens can be revoked and in practice on a LAN the risk of token stealing is probably fairly low!

MyForest, 2020-02-26:
FYI, I know this isn't the solution, but I put this workaround in my app when the AJAX calls don't work.

If it looks like an auth problem it re-directs to the login, otherwise it shows the problem in a non-modal way.

https://github.com/MyForest/mmspheatpump/commit/6baf98efeef563e1173415a607e4e12639f67a16

bwduncan, 2020-03-06:
Looking more closely at this code I think this is just a typo.

```
diff --git a/Modules/feed/feed_controller.php b/Modules/feed/feed_controller.php
index 16207205..c4ce27b6 100644
--- a/Modules/feed/feed_controller.php
+++ b/Modules/feed/feed_controller.php
@@ -144,11 +144,11 @@ function feed_controller()
                     return array('success'=>false, 'message'=> count($missing) .' feeds do not exist');
                 } else {
                     
-                    if ($singular && count($results)==1) {
-                        return $results[0]['data'];
+                    if ($singular && count($result)==1) {
+                        return $result[0]['data'];
                     } else {
-                        return $results;
-                    } 
+                        return $result;
+                    }
                     // @todo: return array for each feed's data 
                     // and a single array for each interval timestamp
                 }
```

That doesn't solve the bug (you just get a different error message which doesn't tell you what the problem is), but the above is surely what was intended by the author.

bwduncan, 2020-03-06:
Oh, please ignore the above patch. It breaks the graphing module completely. Everything returns `{"success": false,"message":"bad parameters"}`

TrystanLea, 2020-03-19:
I've fixed the php error highlighted by @bwduncan and added a automatic redirect to the login page so far on the feeds list page:
https://github.com/emoncms/emoncms/commit/295879278153e85910bdfb740bc6ec87506dd6d9

For the graphs view, is this specifically graphs when embedded in private dashboards?

TrystanLea, 2020-03-19:
I've added the same auto logout to the inputs page as well. Its not as straightforward to add to the dashboards and graph view due to the interaction with public feeds.. I will have to think about that some more.

closed 2023-04-06 by TrystanLea